gdsfactory/gdsfactory #3334: Failed to exclude ports when adding and routing grating couplers

cheng1214, 2024-11-03:
Hi!
@joamatab @sebastian-goeldi 

For example, when I ran code below to exclude 'o1', 'o2' ports in the routing, it raised an error "_ValueError: ports1=4 and ports2=2 must be equal_"

```
import gdsfactory as gf

c = gf.components.crossing()
cc = gf.routing.add_fiber_array(
    component=c,
    grating_coupler=gf.components.grating_coupler_elliptical_te,
    with_loopback=False,
    excluded_ports = ['o1', 'o2']
)
cc.plot()
```

Thanking you!

cheng1214, 2024-11-06:
I just wondered if anyone had an idea how that could be fixed?

sebastian-goeldi, 2024-11-07:
The error you are getting is because something is trying to route or place 4 ports on one side and 2 ports on the other. But I only know that it comes from the routing function or add fiber array function from gdsfactory and I am not familiar with those.

joamatab, 2024-11-10:
Hi Cheng,

thank you for reporting this

See fix
https://github.com/gdsfactory/gdsfactory/pull/3355

after the fix it works well

[image]

closed 2024-11-10 by joamatab