gdsfactory/gdsfactory #3342: clear_cache() causes import_gds() to fail until Python kernel is restarted

karl-johnson, 2024-11-07:
Some background which I am including as I think this is bad practice but I don't know another way: in every layout I make, I put `gf.clear_cache()` at the beginning of the file. Without this, making changes inside functions and running the code again without restarting the kernel typically does not show any of the new changes. My understanding of this is because the cell name is cached, and @gf.cell() does not update the cell name unless function arguments are updated, so the cached cell is used instead of actually calling the function. Is it good practice to use `clear_cache` like this? Is there a way to force gdsfactory to ignore the cache for a specific cell without using `clear_cache`?

With that in mind...

**Describe the bug**
Importing a gds file using `import_gds()`, calling `clear_cache()`, then attempting to import the same file again throws this error:

 ```
File ~\AppData\Roaming\Python\Python311\site-packages\kfactory\kcell.py:2360 in write
    for kcell in (self.kcl[ci] for ci in self.called_cells()):
RuntimeError: Object has been destroyed already in Cell.called_cells
```

**To Reproduce**
This triggers the error:
```
import gdsfactory as gf
gf.import_gds('[PATH TO ANY GDS FILE]')
gf.clear_cache()
gf.import_gds('[PATH TO ANY GDS FILE]')
```
Running this twice (without clearing variables or restarting the kernel inbetween) does as well:
```
import gdsfactory as gf
gf.clear_cache()
gf.import_gds('[PATH TO ANY GDS FILE]')
```
Interestingly, running this twice does *not* trigger the error:
```
import gdsfactory as gf
gf.import_gds('[PATH TO ANY GDS FILE]')
gf.clear_cache()
```

**Expected behavior**
No error when calling `import_gds` after `clear_cache`.

**Suggested fix**
Despite spending an hour or two trying to step through the code in my spare time over the past few days, I haven't been able to understand how the caching works in gdsfactory to determine a fix here. It seems like `clear_cache` leaves behind something which causes import_gds to look at a (cached?) object that has been destroyed, but that's just a guess with no direct evidence.

**Environment (please complete the following information):**
Scripts are being run in the Spyder IDE so variables etc are retained between subsequent runs of the same script.
```
3.11.4 | packaged by Anaconda, Inc. | (main, Jul  5 2023, 13:38:37) [MSC v.1916 64 bit (AMD64)]
C:\Users\Kyle\anaconda3\python.exe
                                    Modules                                    
┌────────────┬───────────────────────────────┬────────────────────────────────┐
│    Package │ version                       │                           Path │
├────────────┼───────────────────────────────┼────────────────────────────────┤
│     python │ 3.11.4 | packaged by          │ C:\Users\Kyle\anaconda3\pytho… │
│            │ Anaconda, Inc. | (main, Jul   │                                │
│            │ 5 2023, 13:38:37) [MSC v.1916 │                                │
│            │ 64 bit (AMD64)]               │                                │
│ gdsfactory │ 8.17.0                        │ C:\Users\Kyle\AppData\Roaming… │
│   gplugins │ 0.3.0                         │ ['C:\\Users\\Kyle\\anaconda3\… │
│        ray │ not installed                 │                                │
│    femwell │ 0.1.9                         │ ['C:\\Users\\Kyle\\anaconda3\… │
│     devsim │ not installed                 │                                │
│     tidy3d │ not installed                 │                                │
│       meep │ not installed                 │                                │
│       meow │ not installed                 │                                │
│     lumapi │ not installed                 │                                │
│        sax │ not installed                 │                                │
└────────────┴───────────────────────────────┴────────────────────────────────┘
```
- [x] I have reviewed the documentation, discussions and issues and found no relevant solution.
- [x] I understand that this is an open-source project, and maintainers may not have the resources to address every issue. I am prepared to contribute by fixing the issue myself or hiring someone to do so, if needed, and accept that the issue may not be resolved otherwise.
- [x] I am using the latest version of GDSFactory, with Python 3.10, 3.11, or 3.12.

sebastian-goeldi, 2024-11-07:
This is because the `import_gds` has a cache decorator. With the clear_cache you are cleaning everything up  in the layout itself but not the cache of the import_gds function.

@joamatab

karl-johnson, 2024-11-07:
Thanks for the insight. Given that a simple workaround working for me is clearing both caches at the beginning of my layout scripts:
``` 
gf.clear_cache()
gf.read.import_gds.cache_clear()
```
I'll leave this issue open as it's still unclear to me whether this is the intentional behavior for gf.clear_cache().

joamatab, 2024-12-25:
Thanks for reporting this Karl

As we added clearing the import_gds function cache as you suggested

closed 2024-12-25 by joamatab