gdsfactory/gdsfactory #3343: Ports silently snap to 90-degree angles

95-martin-orion, 2024-11-07:
**Describe the bug**
A `gf.Port` created with an orientation that is not a multiple of 90 will automatically set its orientation to the next lower multiple of 90. This does not raise an error or produce a warning message.

(Context: I'm trying to set up all-angle routing for my project's gdsfactory integration. It's not immediately clear whether we'll need this behavior, but we used it before switching to gdsfactory.)

**To Reproduce**
```python
p=gf.Port(name="foo", orientation=359, center=(0, 0), width=5, layer=(1, 1))
print(p.orientation)
# 270.0
```

**Expected behavior**
Off-angle ports should either be fully supported or generate an error/warning when created.

**Suggested fix**
Based on recent kfactory changes, it seems like this could be supported with a "virtual port" concept? Otherwise raising an error for invalid port orientations would be very helpful for debugging.

**Environment (please complete the following information):**

- [x] I have reviewed the documentation, discussions and issues and found no relevant solution.
    - [This page](https://gdsfactory.github.io/gdsfactory/notebooks/12_grid_snapping.html#warning-manhattan-orientations) warns against non-manhattan rotations and suggests that the feature exists, but doesn't actually indicate how to use it.
- [x] I understand that this is an open-source project, and maintainers may not have the resources to address every issue. I am prepared to contribute by fixing the issue myself or hiring someone to do so, if needed, and accept that the issue may not be resolved otherwise.
- [x] I am using the latest version of GDSFactory, with Python 3.10, 3.11, or 3.12.

Please provide the output for the following code:

<details><summary>Output</summary>

```python
>>> import sys
>>> print(sys.version)
3.12.2 (main, Feb 10 2024, 20:26:15) [GCC 12.2.0]
>>> print(sys.executable)
/usr/local/google/home/martinop/.virtualenvs/pyle-gf8/bin/python
>>> import gdsfactory as gf_plugins()
>>> gf.config.print_version_plugins()
                 Modules                  
┏━━━━━━━━━━━━┳━━━━━━━━━━━━━┳━━━━━━━━━━━━━┓
┃    Package ┃ version     ┃        Path ┃
┡━━━━━━━━━━━━╇━━━━━━━━━━━━━╇━━━━━━━━━━━━━┩
│     python │ 3.12.2      │ [REDACTED]  │
│            │ (main, Feb  │             │
│            │ 10 2024,    │             │
│            │ 20:26:15)   │             │
│            │ [GCC        │             │
│            │ 12.2.0]     │             │
│ gdsfactory │ 8.17.0      │ [REDACTED]  │
│   gplugins │ not         │             │
│            │ installed   │             │
│        ray │ not         │             │
│            │ installed   │             │
│    femwell │ not         │             │
│            │ installed   │             │
│     devsim │ not         │             │
│            │ installed   │             │
│     tidy3d │ not         │             │
│            │ installed   │             │
│       meep │ not         │             │
│            │ installed   │             │
│       meow │ not         │             │
│            │ installed   │             │
│     lumapi │ not         │             │
│            │ installed   │             │
│        sax │ not         │             │
│            │ installed   │             │
└────────────┴─────────────┴─────────────┘
```

</details>

sebastian-goeldi, 2024-11-07:
Culprit is here:

https://github.com/gdsfactory/gdsfactory/blob/7b209cd9d36190aef02825e528d6302996522a94/gdsfactory/port.py#L158

This snaps it because it uses trans and not dcplx_trans. If it uses dcplx_trans, it also needs to use dwidth as a kwarg

joamatab, 2024-11-15:
Thank you for reporting this Orion,

See fix here
https://github.com/gdsfactory/gdsfactory/pull/3386

closed 2024-11-15 by joamatab