gdsfactory/gdsfactory #3350: improve graphviz

joamatab, 2024-11-09:
- [ ] use corner arrows instead of straight arrows 

@Jelly298

joamatab, 2024-11-09:
using splines 'ortho' i got it working but the edges don't connect to the ports

https://github.com/gdsfactory/gdsfactory/pull/3351

[image]

Jelly298, 2024-11-11:
I've reproduced the issue. Also found a similar issue here. 

https://stackoverflow.com/questions/75953069/graphviz-edges-with-ortho-lines-not-aligning-with-ports

Working on the fix

joamatab, 2024-11-11:
Awesome, the new schematics look beautiful thanks to your PR, [see](https://gdsfactory.github.io/gdsfactory/notebooks/10_schematic.html)

closed 2024-11-25 by joamatab