gdsfactory/gdsfactory #3369: Many tests failing

fedeinthemix, 2024-11-12:
**Describe the bug**
What's the bug?
I'm packaging `gdsfactory` for [NixOS](https://nixos.org/). It compiles and simple examples from the tutorial do work. However, we normally prefer to run the tests of packages. When I run them, I find that many tests fail (287!), with a large fraction because they can't find GDS files.

Is the test suite supposed to pass all tests? (NixOS is unconventional and it could be that they are failing for some reason I don't currently understand.)

Here the the build log

[gdsfactory.log](https://github.com/user-attachments/files/17716834/gdsfactory.log)

**Expected behavior**
I'd expect all tests to pass.

**Suggested fix**
How could we fix the bug? As an open-source project, we welcome your suggestions and PRs.

**Environment (please complete the following information):**

- [x] I have reviewed the documentation, discussions and issues and found no relevant solution.
- [x] I understand that this is an open-source project, and maintainers may not have the resources to address every issue. I am prepared to contribute by fixing the issue myself or hiring someone to do so, if needed, and accept that the issue may not be resolved otherwise.
- [x] I am using the latest version of GDSFactory, with Python 3.10, 3.11, or 3.12.

Please provide the output for the following code:

```python
import sys
print(sys.version)
print(sys.executable)

import gdsfactory as gf
gf.config.print_version_plugins()

3.12.7 (main, Oct  1 2024, 02:05:46) [GCC 13.3.0]
/nix/store/mzq6hm06nigc2p9n6xdnz99hxhqx86jv-python3-3.12.7-env/bin/python3.12
                                      Modules                                       
┏━━━━━━━━━━━━┳━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━┳━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━┓
┃    Package ┃ version                          ┃                             Path ┃
┡━━━━━━━━━━━━╇━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━╇━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━┩
│     python │ 3.12.7 (main, Oct  1 2024,       │ /nix/store/mzq6hm06nigc2p9n6xdn… │
│            │ 02:05:46) [GCC 13.3.0]           │                                  │
│ gdsfactory │ 8.18.1                           │ /home/beffa/src/git/nixpkgs/res… │
│   gplugins │ not installed                    │                                  │
│        ray │ not installed                    │                                  │
│    femwell │ not installed                    │                                  │
│     devsim │ not installed                    │                                  │
│     tidy3d │ not installed                    │                                  │
│       meep │ not installed                    │                                  │
│       meow │ not installed                    │                                  │
│     lumapi │ not installed                    │                                  │
│        sax │ not installed                    │                                  │
└────────────┴──────────────────────────────────┴──────────────────────────────────┘

```

sebastian-goeldi, 2024-11-12:
You need to have the reference gds files, otherwise the tests fail due to missing reference data.

The tests are doing it with the following lines from the Makefile

https://github.com/gdsfactory/gdsfactory/blob/main/Makefile#L22-L23

fedeinthemix, 2024-11-12:
OK. thanks!

closed 2024-11-12 by fedeinthemix