gdsfactory/gdsfactory #3373: Issues with KLayout version and multiple cell copies in GenISys BEAMER

danielpeace, 2024-11-13:
We recently upgraded from version 7 to 8 and have encountered a few issues with our gds files. Initially, we weren't able to open the generated gds files with older version of klayout. Appears the gds files only work with latest klayout version. 

Secondly, when importing our GDS files into GenISys BEAMER software in order to process the files for EBL we're seeing two copies of each of the cells which make up the gds. One cell is shown as expected in the right locations. But we also see copies of all the components which make up those cells placed at (0,0). Interestingly these cell copies placed at (0,0) don't show up in klayout.

Given this it seems there is maybe a bug with the handling of cells which is related to both the klayout issue and the multiple cell copies.

GDS in KLayout
[image]

View in BEAMER
[image]

Code below was used to generate the gds file. Running gdsfactory version 8.8.5 and Python 3.12.7 on Windows WSL with Ubuntu

```python
import gdsfactory as gf

@gf.cell
def spiral_gc(**kwargs):
    """Returns spiral with Grating Couplers."""
    c = gf.components.spiral_double(**kwargs)
    c = gf.routing.add_fiber_array(c)
    c.info["doe"] = "spirals_sc"  # strip Cband spirals
    c.info["measurement"] = "optical_spectrum"
    c.info["analysis"] = "[power_envelope]"
    c.info["analysis_parameters"] = "[]"
    c.info["ports_optical"] = 4
    c.info["ports_electrical"] = 0
    c.info.update(kwargs)
    return c

if __name__ == "__main__":
    sweep = [spiral_gc(number_of_loops=n) for n in [3,5,7]]
    m = gf.pack(sweep)
    c = m[0]

    c.write_gds()
```

sebastian-goeldi, 2024-11-13:
Pretty sure this is BEAMER mishandling or partially misinterpreting klayout metadata.

please try with `c.write_gds(with_metadata=False)` and open that in BEAMER.

Wrt klayout, yes that's partially true. It works with anything > 0.29 for sure, and I am reasonably sure it works with anything released in the last year, i.e. >= 0.28.13

danielpeace, 2024-11-13:
Hi Sebastian, thanks for the reply.

Looks like you're correct, without the metadata it imports correctly into BEAMER.

closed 2024-11-14 by joamatab