gdsfactory/gdsfactory #3433: gf.Component.remove_layers() related question

rgiroud, 2024-11-26:
Hi, 

I've come across a strange behavior from the function 'remove_layers()', I was wondering if this was intended and I miss-understood how to use it or if this was a bug.

I'm trying to copy a flattened Component (using gf.Component.dup()) and remap/remove layers in the copy. 
- 'remap_layers()' works exactly as I expected, leaving the original Component untouched. 
- However, 'remove_layers()' removes the layer from the original Component.

Is this the expected result? 

Here is a piece of code to reproduce: 

```
comp = gf.Component()
r1 = gf.components.rectangle(size=(1, 15), layer=(123, 0), centered=True)
comp << r1
r2 = gf.components.rectangle(size=(3, 17), layer=(124, 0), centered=True)
comp << r2
comp.flatten()

copy = comp.dup()
copy.remap_layers(layer_map={(123, 0): (13, 0)})
```

[image]

By adding the following line: 
`copy.remove_layers(layers=[(124, 0)])`

the output becomes: 

[image]

What I would expect is the outside rectangle to be removed from the copy but not from the original Component since they are supposedly intrinsically different Components. (I made sure to flatten 'comp' otherwise, this would not necessarily be the same situation).

<details>
  <summary>Code to generate the images</summary>
  
```
main = gf.Component()
main << comp
label1 = gf.components.text(text="Original", size=1, layer=(123, 0))
ref_lab = main << label1
ref_lab.drotate(90)
ref_lab.dmove(origin=ref_lab.dsize_info.se, destination=main.dsize_info.sw)
ref = main << copy
ref.dmovex(5)
label1 = gf.components.text(text="Copy", size=1, layer=(123, 0))
ref_lab = main << label1
ref_lab.drotate(90)
ref_lab.dmove(origin=ref_lab.dsize_info.se, destination=ref.dsize_info.sw)
main.show()

```
 
</details>

Thanks in advance for the answer :)
Robin

rgiroud, 2024-11-26:
Oh thanks for the answer, I didn't notice the `recursive` was `True` by default in `remove_layers()` !  This effectively solves my problem.

sebastian-goeldi, 2024-11-26:
`copy_tree(source_cell)` is what you would want for that https://klayout.de/doc/code/class_Cell.html#method41

rgiroud, 2024-11-27:
Thanks a lot for the hint!

joamatab, 2024-11-27:
How would you do a recursive delete that is not global to the layout?

sebastian-goeldi, 2024-11-28:
`[c.kcl[ci].shapes(layer).clear() for ci in cell.called_cells() for layer in removed_layers]`

joamatab, 2024-11-28:
Awesome!

See fix [here](https://github.com/gdsfactory/gdsfactory/pull/3439)

closed 2024-11-28 by joamatab

rgiroud, 2024-11-28:
Works nicely now. Thanks !