gdsfactory/gdsfactory #3434: Variable width with hard-coded points

Arka-noid, 2024-11-27:
**Is your feature request related to a problem? Please describe.**
Variable width cross-sections require a parametrized function from [0, 1] with equally spaced points.
To define piece-wise functions, it is necessary to combine and put in series several pieces of waveguides/cross-sections.
Instead of using component_sequence, it would be desirable to define a profile with [x, y] points

**Describe the solution you'd like**
Not all variable width cases can be covered with parametrized function unless someone does an interpolation fit function with some points.
Example: multi-section linear taper. I'd like to easily define a sequence of transitions all with linear slope (or even non-linear) with different lengths. 

But even using parametrized functions, it is not always desirable to have equally spaced points.

- [x] I understand that, as an open-source project, the maintainers may have limited capacity to address every request. I am open to contributing by developing the feature myself or hiring someone to implement it (contact@gdsfactory.com). Otherwise, I understand that this feature may not be prioritized.

joamatab, 2024-11-27:
we could add something like this

```python

import gdsfactory as gf
import numpy as np

def piese_wise_waveguide(x, y, layer="WG", sections=None, port_names=("o1", "o2"), name='core', **kwargs):

    def width_function(): 
        return np.array(y)

    P = gf.Path()
    P.points = np.array([(xi, 0.) for xi in x])

    sections = sections or []
    sections = list(sections)
    sections+= gf.Section(name=name, width=0, width_function=width_function, offset=0, layer=layer, port_names=port_names, **kwargs)
    cross_section = gf.CrossSection(sections=sections)

    # Extrude the Path to create the Component
    c = gf.path.extrude(P, cross_section=cross_section)
    return c
    
c = piese_wise_waveguide([0, 4., 20, 40], [1, 2, 0.15, 0.4])
c.add_polygon([(0, 0), (0, 1), (40, 1), (40, 0)], layer="WG")
c.show()

```

Arka-noid, 2024-11-28:
This could be an option, indeed, I did not include the possibility to insert the other options. 
I think in some cases it could be useful to have a waveguide defined like that. 

In other cases you may import a profile that you simulate that it has no obvious parametrization. Maybe you also want to follow a curved path (e.g. a bend that you simulated with inverse design).

MatthewMckee4, 2024-12-28:
Here's a proper write up with types etc, also the ability to supply a path. How does this look?

```python
from collections.abc import Sequence
from typing import Any

import numpy as np
import numpy.typing as npt

import gdsfactory as gf
from gdsfactory.component import Component
from gdsfactory.cross_section import Section
from gdsfactory.path import Path
from gdsfactory.typings import LayerSpec

def piece_wise_waveguide(
    x: Sequence[float] | Path,
    y: Sequence[float],
    layer: LayerSpec = "WG",
    sections: Sequence[Section] | None = None,
    port_names: tuple[str | None, str | None] = ("o1", "o2"),
    name: str = "core",
    **kwargs: Any,
) -> Component:
    """Create a component with a piecewise waveguide.

    Args:
        x (Union[Sequence[float], gf.Path]): x coordinates of the piecewise function or a custom path.
        y (Sequence[float]): y coordinates of the piecewise function
        layer (LayerSpec): layer to extrude.
        sections (Sequence[Section] | None): sections to extrude.
        port_names (tuple[str | None, str | None]): port names.
        name (str, optional): name of the component.
        **kwargs: additional keyword arguments for the Section.
    """
    if isinstance(x, Sequence) and len(x) != len(y):
        raise ValueError("x and y must have the same length")

    def width_function(_: float) -> npt.NDArray[np.floating[Any]]:
        return np.array(y)

    if isinstance(x, gf.Path):
        p = x
    else:
        p = gf.Path()
        p.points = np.array([(xi, 0.0) for xi in x])

    section_list = list(sections or [])
    section_list.append(
        Section(
            name=name,
            width=0,
            width_function=width_function,
            offset=0,
            layer=layer,
            port_names=port_names,
            **kwargs,
        )
    )
    cross_section = gf.CrossSection(sections=tuple(section_list))

    c = gf.path.extrude(p, cross_section=cross_section)
    return c
```

closed 2024-12-29 by joamatab