gdsfactory/gdsfactory #3435: Circuit without overlapping components gets plotted with overlapping components

ferranAX, 2024-11-27:
**Describe the bug**
When writing the connections in a netlist I noticed an error. To specify connections I define a map like: {port1: port2} to indicate that port1 is connected to port2. To my understanding this should be equivalent to specifying {port2: port1}, that is, switching the key and value. However, when plotting two identical circuits, but one with {port1: port2} and the other with {port2: port1}, one of them is correct, but the other overlaps components and connects them to the wrong ports.

**To Reproduce**
```python
import gdsfactory as gf
import yaml

netlist_1 = {
    'instances': {
        'y_branch': {'component': 'mmi1x2'},
        'bend1': {'component': 'bend_euler', 'settings': {'angle': 45}},
        'bend2': {'component': 'bend_euler', 'settings': {'angle': -45}}},
    'connections': {'bend1,o1': 'y_branch,o2', 
                    'bend2,o1': 'y_branch,o3'},
    'ports': {'input': 'y_branch,o1', 
              'output1': 'bend2,o2',
              'output2': 'bend1,o2'}}
file_path_1 = "test_image1.png"

netlist_2 = netlist_1.copy()
#swap key value pairs
netlist_2["connections"] = {value: key for key,value in netlist_1["connections"].items()}
file_path_2 = "test_image2.png"

gf.read.from_yaml(yaml.safe_dump(netlist_1)).plot()
```
[image]
```python
gf.read.from_yaml(yaml.safe_dump(netlist2)).plot()
```
[image]

**Expected behavior**
The plots should be the same. 

**Suggested fix**
I have no clue what may be causing this :(

**Environment (please complete the following information):**

- [x] I have reviewed the documentation, discussions and issues and found no relevant solution.
- [x] I understand that this is an open-source project, and maintainers may not have the resources to address every issue. I am prepared to contribute by fixing the issue myself or hiring someone to do so, if needed, and accept that the issue may not be resolved otherwise.
- [x] I am using the latest version of GDSFactory, with Python 3.10, 3.11, or 3.12.

The output of the following code
```python
import sys
print(sys.version)
print(sys.executable)

import gdsfactory as gf
gf.config.print_version_plugins()
```
is
3.11.10 (main, Oct  3 2024, 02:26:51) [Clang 14.0.6 ]
/opt/anaconda3/envs/env1/bin/python
/opt/anaconda3/envs/env1/lib/python3.11/site-packages/sax/backends/__init__.py:58: UserWarning: klujax not found. Please install klujax for better performance during circuit evaluation!
  warnings.warn(
                                                      Modules                                                      
┏━━━━━━━━━━━━┳━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━┳━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━┓
┃    Package ┃ version                                         ┃                                             Path ┃
┡━━━━━━━━━━━━╇━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━╇━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━┩
│     python │ 3.11.10 (main, Oct  3 2024, 02:26:51) [Clang    │          /opt/anaconda3/envs/env1/bin/python │
│            │ 14.0.6 ]                                        │                                                  │
│ gdsfactory │ 8.20.0                                          │ /opt/anaconda3/envs/env1/lib/python3.11/sit… │
│   gplugins │ 0.14.0                                          │ ['/opt/anaconda3/envs/env1/lib/python3.11/s… │
│        ray │ not installed                                   │                                                  │
│    femwell │ not installed                                   │                                      
            │
│     devsim │ not installed                                   │                                                  │
│     tidy3d │ 2.5.2                                           │ ['/opt/anaconda3/envs/env1/lib/python3.11/s… │
│       meep │ not installed                                   │                                                  │
│       meow │ not installed                                   │                                                  │
│     lumapi │ not installed                                   │                                                  │
│        sax │ 0.13.3                                          │ ['/opt/anaconda3/envs/env1/lib/python3.11/s…/lib/python3.11/s…

StephenAnderson89, 2024-11-27:
Some details in from_yaml.py:
gf.read.from_yaml calls _place_and_connect, which calls _graph_roots and places components in a loop for each root in the graph.
For the netlist_1 case, _graph_roots returns one root, 'y_branch', while for the netlist_2 case, _graph_roots returns two roots, ['bend1', 'bend2'].

The yaml string in this example does not supply placements, only connections, so when _place_and_connect loops through each graph root, it places the "roots" at the default Placement (x=0, y=0, etc.).  
For the netlist_2 case, this places both bend1 and bend2 at 0,0. Then, other components are transformed using .connect, to move the reference (in this case, the y branch). It is standard gdsfactory syntax that .connect() transforms a reference but does not maintain that relationship between the two components if one is moved again. 

If the following logger is added at the end of the loop in _place_and_connect:
                    from gdsfactory import logger
                    logger.debug(f'connected port {p1} of {refs[i1].name} to port {p2} of {refs[i2].name}')
You get
Netlist_1 :

2024-11-27 11:30:49.251 | DEBUG    | gdsfactory.read.from_yaml:_place_and_connect:866 - connected port o1 of bend1 to port o2 of y_branch
2024-11-27 11:30:49.252 | DEBUG    | gdsfactory.read.from_yaml:_place_and_connect:866 - connected port o1 of bend2 to port o3 of y_branch
 
Netlist_2:

2024-11-27 11:39:19.059 | DEBUG    | gdsfactory.read.from_yaml:_place_and_connect:866 - connected port o2 of y_branch to port o1 of bend1
2024-11-27 11:39:19.060 | DEBUG    | gdsfactory.read.from_yaml:_place_and_connect:866 - connected port o3 of y_branch to port o1 of bend2

in the case of Netlist_2:
This second connection overrides the first (standard gdsfactory behavior), moving the Y_branch port o3 to be coincident with the port o1 of bend2. bend1 and bend2 are co-incident because they are both treated as roots of the directed graph generated from the netlist, generated from the yaml string. bend1 and bend2 are co-incident because they are both treated as roots of the directed graph generated from the netlist, generated from the yaml string. The placement of the roots happens before the placement of the references.

joamatab, 2024-11-27:
It seems the issue arises because `{src: dst}` is not equivalent to `{dst: src}` in the netlist. The directionality of the connection matters since it determines how components are positioned and linked.

When specifying `{port1: port2}`, the system likely interprets `port1` as the source (`src`) and `port2` as the destination (`dst`). Reversing this to `{port2: port1}` flips their roles, which may result in:

- Incorrect placement of components
- Overlapping components
- Wrong port connections

This distinction is crucial for consistent netlist generation and visualization. To avoid such mismatches, consider standardizing the direction of connections in your map (e.g., always `{src: dst}`) and ensuring your plotting logic respects this convention.

sebastian-goeldi, 2024-11-28:
Yes, exactly. The problem is that a connection in gdsfactory is not really "fusing" components, it's merely a placement statement. So if you swap them, it first tells the mmi to first move and rotate inna way that allows it to be connected to the first bend. The second statement tells it to do that again for the second bend.

As you can see from this, neither bend is ever touched.

@joamatab might be a good idea to give a warning or error if an instanced is moved more than once. What would happen if the mmi was in the placement section as well? What would take precedence, placement or connection? That would be very unclear from the yaml alone.

ferranAX, 2024-11-28:
Thanks a lot for the response @StephenAnderson89 :) I now understand what was happening!

@joamatab I must say however that it is quite counterintuitive that specifying {src: dst} will generate the directed edge dst -> src. So the actual graph roots were what I thought were its leaves, I had not taken into account that you built your graph specifying that dst depends on src.

The swap of src,dst to build the edge is done in line 965 of from_yaml, for the record in case it was not intentional:

```python
def _graph_connect(g: nx.DiGraph, i1: str, i2: str) -> None:
    g.add_edge(i2, i1)  # type: ignore
```

Overall I agree with @sebastian-goeldi in that some warning or specification would be nice :)

closed 2024-11-28 by ferranAX