gdsfactory/gdsfactory #3497: support more complex routing

joamatab, 2024-12-14:
```python

import gdsfactory as gf

c = gf.Component()
columns = 2
ptop = c << gf.components.pad_array(columns=columns, port_orientation=270)
pbot = c << gf.components.pad_array(port_orientation=270, columns=columns)
ptop.movex(300)
ptop.movey(300)

obstacle = c << gf.c.rectangle(size=(300, 100), layer="M3")
obstacle.ymin = pbot.ymax - 10
obstacle.xmin = pbot.xmax - 10

for port1, port2 in zip(pbot.ports, ptop.ports):
    routes = gf.routing.route_astar(
        c,
        port1,
        port2,
        cross_section="metal_routing",
        radius=10,
        avoid_layers=("M3",),
        bend=gf.c.bend_euler,
    )

c.show()

```

Deepa0325, 2025-02-03:
> import gdsfactory as gf
> 
> c = gf.Component()
> columns = 2
> ptop = c << gf.components.pad_array(columns=columns, port_orientation=270)
> pbot = c << gf.components.pad_array(port_orientation=270, columns=columns)
> ptop.movex(300)
> ptop.movey(300)
> 
> obstacle = c << gf.c.rectangle(size=(300, 100), layer="M3")
> obstacle.ymin = pbot.ymax - 10
> obstacle.xmin = pbot.xmax - 10
> 
> 
> for port1, port2 in zip(pbot.ports, ptop.ports):
>     routes = gf.routing.route_astar(
>         c,
>         port1,
>         port2,
>         cross_section="metal_routing",
>         radius=10,
>         avoid_layers=("M3",),
>         bend=gf.c.bend_euler,
>     )
> 
> c.show()
> [@Jelly298](https://github.com/Jelly298)

I have an error running this code.
I'm using python 3.12.4

ModuleNotFoundError                       Traceback (most recent call last)
Cell In[3], [line 1](vscode-notebook-cell:?execution_count=3&line=1)
----> [1](vscode-notebook-cell:?execution_count=3&line=1) import gdsfactory as gf
      [3](vscode-notebook-cell:?execution_count=3&line=3) c = gf.Component()
      [4](vscode-notebook-cell:?execution_count=3&line=4) columns = 2

ModuleNotFoundError: No module named 'gdsfactory'

Jelly298, 2025-02-04:
@Deepa0325 I could not reproduce your error. Maybe "ModuleNotFoundError" is caused by missing installations of gdsfactory? Have you tried reinstalling gdsfactory?

Deepa0325, 2025-02-04:
> [@Deepa0325](https://github.com/Deepa0325) I could not reproduce your error. Maybe "ModuleNotFoundError" is caused by missing installations of gdsfactory? Have you tried reinstalling gdsfactory?

I have used 'Visual Studio code' to run code integrated with klayout.
Klayout 0.29.11 - gdsfactory 8.31.2.