gdsfactory/gdsfactory #3528: Schematic.from_component not working

MatthewMckee4, 2024-12-24:
**Describe the bug**
In the function `component.to_yaml()` does not supply any arguments:
```bash
    n = component.to_yaml()
        ^^^^^^^^^^^^^^^^^^^
TypeError: KCell.to_yaml() missing 2 required positional arguments: 'representer' and 'node'
```

MatthewMckee4, 2024-12-24:
I can figure out the type of representer, but im not sure the type of node, seems like some kfactory type

sebastian-goeldi, 2024-12-24:
This does not directly work this way.

NB: this is not the gdsfactory netlist representation. It's a kfactory yaml version.

To use this version, you need to use `kf.placer.cells_from_yaml` or the `to_yaml` version. The to_yaml in KCell is purely theee for ruamel.yaml to represent the cell in yaml or convert back. Tbh I haven't recently tested it yet, so it's possible it will fails for some info/settings.

joamatab, 2024-12-28:
Hi Matthew, 

yes, this is a bit confusing, 

you need to use `n = Component.get_netlist()`

@tvt173

closed 2025-01-05 by joamatab