gdsfactory/gdsfactory #3529: gf.read.import_gds() not working

Happy-feet0711, 2024-12-24:
**Describe the bug**
Hi guys, I tried to use import_gds() to open the gds file I exported from cadence but failed. But when I used import_gds() to open the gds file I writed by gdsfactory, it succeeded. 
Thank you for the help.

**To Reproduce**

```
import gdsfactory as gf
ly = gf.read.import_gds(
    path,
)
```

**Error Message**
```
---------------------------------------------------------------------------
KeyError                                  Traceback (most recent call last)
Cell In[25], line 2
      1 # import layout
----> 2 ly = gf.read.import_gds(
      3     path,
      4 )
      6 # ly.show()

File C:\Python\Python311\Lib\site-packages\gdsfactory\read\import_gds.py:39, in import_gds(gdspath, cellname, post_process, rename_duplicated_cells, **kwargs)
     37 options = kf.kcell.load_layout_options()
     38 options.warn_level = 0
---> 39 temp_kcl.read(gdspath, options=options)
     40 cellname = cellname or temp_kcl.top_cell().name
     41 kcell = temp_kcl[cellname]

File C:\Python\Python311\Lib\site-packages\kfactory\kcell.py:5121, in KCLayout.read(self, filename, options, register_cells, test_merge, update_kcl_meta_data, meta_format)
   5119     for c in sorted(new_cells, key=lambda _c: _c.hierarchy_levels()):
   5120         kc = KCell(kdb_cell=c, kcl=self)
-> 5121         kc.get_meta_data(
   5122             meta_format=meta_format,
   5123         )
   5125 for c in load_cells & cells:
   5126     kc = self.kcells[c.cell_index()]

File C:\Python\Python311\Lib\site-packages\kfactory\kcell.py:2792, in KCell.get_meta_data(self, meta_format)
   2787 _trans: kdb.Trans | None = _v.get("trans")
   2788 if _trans is not None:
   2789     self.create_port(
   2790         name=_v.get("name"),
   2791         trans=_trans,
-> 2792         cross_section=self.kcl.get_cross_section(
   2793             _v["cross_section"]
   2794         ),
   2795         port_type=_v["port_type"],
   2796     )
   2797 else:
   2798     self.create_port(
   2799         name=_v.get("name"),
   2800         dcplx_trans=_v["dcplx_trans"],
   (...)
   2804         port_type=_v["port_type"],
   2805     )

File C:\Python\Python311\Lib\site-packages\kfactory\kcell.py:5285, in KCLayout.get_cross_section(self, cross_section)
   5277 def get_cross_section(
   5278     self,
   5279     cross_section: str
   (...)
   5282     | DSymmetricalCrossSection,
   5283 ) -> SymmetricalCrossSection:
   5284     """Get a cross section by name or specification."""
-> 5285     return self.cross_sections.get_cross_section(cross_section)

File C:\Python\Python311\Lib\site-packages\kfactory\kcell.py:3698, in CrossSectionModel.get_cross_section(self, cross_section)
   3689         return self.get_cross_section(
   3690             CrossSectionSpec(
   3691                 sections=cross_section.enclosure.model_dump()["sections"],
   (...)
   3695             )
   3696         )
   3697 if isinstance(cross_section, str):
-> 3698     return self.cross_sections[cross_section]
   3699 elif isinstance(cross_section, DSymmetricalCrossSection):
   3700     cross_section = cross_section.to_itype(self.kcl)

KeyError: 'cfffc3a0_660'
```

**Environment:**
```python
3.11.2 (tags/v3.11.2:878ead1, Feb  7 2023, 16:38:35) [MSC v.1934 64 bit (AMD64)]
C:\Python\Python311\python.exe
                                                      Modules                                                      
┏━━━━━━━━━━━━┳━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━┳━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━┓
┃    Package ┃ version                                          ┃                                            Path ┃
┡━━━━━━━━━━━━╇━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━╇━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━┩
│     python │ 3.11.2 (tags/v3.11.2:878ead1, Feb  7 2023,       │                  C:\Python\Python311\python.exe │
│            │ 16:38:35) [MSC v.1934 64 bit (AMD64)]            │                                                 │
│ gdsfactory │ 8.23.0                                           │ C:\Python\Python311\Lib\site-packages\gdsfacto… │
│   gplugins │ not installed                                    │                                                 │
│        ray │ not installed                                    │                                                 │
│    femwell │ not installed                                    │                                                 │
│     devsim │ not installed                                    │                                                 │
│     tidy3d │ not installed                                    │                                                 │
│       meep │ not installed                                    │                                                 │
│       meow │ not installed                                    │                                                 │
│     lumapi │ not installed                                    │                                                 │
│        sax │ not installed                                    │                                                 │
└────────────┴──────────────────────────────────────────────────┴─────────────────────────────────────────────────┘
```

joamatab, 2024-12-24:
Can you `pip install gdsfactory --upgrade`?

If you can share the GDS file it would help us to reproduce your issue

Happy-feet0711, 2024-12-25:
> Can you `pip install gdsfactory --upgrade`?
> 
> If you can share the GDS file it would help us to reproduce your issue

I update the gdsfactory but it did't work.

I tried several times and I need to correct myself. I can use import_gds() to open the gds file exported from cadence. However, if I imported the gds file created by gdsfactory into the cadence gds file, the above error could happen when I try to open the modified gds file again. 

Maybe you can take the attached files for test.
[test.zip](https://github.com/user-attachments/files/18243051/test.zip)

By the way, why is an oas file auto-created after running the import_gds()?

joamatab, 2024-12-25:
can you provide all the code to reproduce? What's the path that gives you the error?

Can you also show the output of this?

```python
import sys
print(sys.version)
print(sys.executable)

import gdsfactory as gf
gf.config.print_version_plugins()
```

joamatab, 2024-12-25:
I can see the error when doing this

```python
import gdsfactory as gf

path = 'test.gds'
ly = gf.read.import_gds(path)
```

@sebastian-goeldi 

any ideas on what could be going on?

sebastian-goeldi, 2024-12-25:
I have some idea what's going on. I think the ports somehow survive the modification in cadence, but the cross sections for the ports don't.

To solve it, export from gdsfactory without meta infos (i.e. `write_gds(with_meta_data=False)`)

Not sure why Cadence keeps part of the info, but not all.

Happy-feet0711, 2024-12-25:
> I have some idea what's going on. I think the ports somehow survive the modification in cadence, but the cross sections for the ports don't.
> 
> To solve it, export from gdsfactory without meta infos (i.e. `write_gds(with_meta_data=False)`)
> 
> Not sure why Cadence keeps part of the info, but not all.

It works. Thank you guys :D

closed 2024-12-25 by Happy-feet0711