gdsfactory/gdsfactory #3553: Modification of a component returned from a function wrapped in gf.cell persists

MatthewMckee4, 2024-12-29:
**Describe the bug**
modification of a component returned from a function wrapped in gf.cell persists when the component is returned from the function again.

**To Reproduce**
Not sure if my description makes sense so hopefully this will suffice.
In add_padding, the function component.add_polygon is called, this on its own maybe seems okay, but the snippet below shows that the "immutability" of the component can cause some issues
```python
c = bend_circular()
print(len(c.get_polygons())) # 1

c = gf.add_padding(c, default=10)
print(len(c.get_polygons())) # 2

d = bend_circular()
print(len(d.get_polygons())) # 2
```
This shows that when we get bend_circular again, it still has the padding polygon.
**Expected behavior**
There is probably some discussion to be made about what the "proper" functionality should be here.
- Should gf.get_component(Component) return a copy to the component
- Should Component.add_polygon return a new component
- Should Component.add_polygon not add polygons to a Component if it is locked

**Suggested fix**
To be honest im not sure about this. Would appreciate some help.

MatthewMckee4, 2024-12-29:
Of these:
- Should gf.get_component(Component) return a copy to the component
- Should Component.add_polygon return a new component
- Should Component.add_polygon not add polygons to a Component if it is locked
I think the first option is not what we want, but i think a combination of the other two is where we could start.

In the kfactory KCell code, i cant really see a way to add a polygon, other than self.shapes(_layer).insert(polygon) (which is used in Component.add_polygon. Perhaps we could start there to make a proper function to add a polygon, which can check if the cell is locked etc.

joamatab, 2024-12-29:
Nice catch, 

once a cell is cached we should disable modification, take a look at the PR

That PR encourages the proper usage of cells where we disallow any modification of the cell