gdsfactory/gdsfactory #3588: extend ports has some issues

joamatab, 2025-01-11:
extend ports doesn't keep slab

```
    c = gf.c.mmi1x2(cross_section='rib')
    c = extend_ports(component=c)
    c.pprint_ports()
    c.show()
```

[image]

sebastian-goeldi, 2025-01-11:
Ports do have crosssections, but not gdsfactory crosssections

joamatab, 2025-01-11:
would `port.info['cross_section'] = 'rib'` be a workaround?

sebastian-goeldi, 2025-01-11:
I guess so. Or we just eliminate the crosssection incompatibilities, namely asymmetrical crosssections and objects along path crosssections

joamatab, 2025-01-11:
that could work too but seems like it would require some more work

sebastian-goeldi, 2025-01-11:
Yes. But it is probably worth it. This would also enable the crosssections being available in the gds

closed 2025-01-14 by joamatab