gdsfactory/gdsfactory #3597: can't update to latest kfactory

joamatab, 2025-01-13:
Im trying to update one PDK to the latest GDFactory and has some issues

This is an example i made to reproduce the issue

```python
import gdsfactory as gf 
from pydantic import BaseModel
from pydantic.dataclasses import ConfigDict

s = gf.Section(width=0.5, offset=0.5, layer='WG')
xs = gf.CrossSection(sections=[s]) # type: ignore

class CrossSectionSettings(BaseModel):
    """Settings."""
    width: float = 0.9
    via_component: gf.Component= gf.c.via()
    model_config = ConfigDict(arbitrary_types_allowed=True, frozen=True)

def xs_custom(settings):
    return gf.cross_section.strip(width=settings.width)

if __name__ == "__main__": 
    settings = CrossSectionSettings()
    x = xs_custom(settings)
    c = gf.c.straight(cross_section=xs)

```

@MatthewMckee4 
@sebastian-goeldi

MatthewMckee4, 2025-01-14:
Will look into it tomorrow

MatthewMckee4, 2025-01-14:
Why do you create 'x' and not use it, only thing i can think is that maybe your supposed to be doing
```python
   c = gf.c.straight(cross_section=x)
```

joamatab, 2025-01-14:
sorry, this is the code

```python
import gdsfactory as gf 
from pydantic import BaseModel
from pydantic.dataclasses import ConfigDict

class CrossSectionSettings(BaseModel):
    """Settings."""
    width: float = 0.9
    via_component: gf.Component= gf.c.via() # This is the problem
    model_config = ConfigDict(arbitrary_types_allowed=True, frozen=True)

def xs_custom(settings):
    return gf.cross_section.strip(width=settings.width)

if __name__ == "__main__": 
    settings = CrossSectionSettings()
    x = xs_custom(settings)
    c = gf.c.straight(cross_section=x)

```

joamatab, 2025-01-14:
it was failing with  pydantic-2.9.2

after updating pydantic i no longer have the issue

closed 2025-01-14 by joamatab