gdsfactory/gdsfactory #3603: Bug in Component.offset()

MatthewMckee4, 2025-01-14:
Just trying to test the Component functions aswell as ComponentAllAngle, ive found this bug, when calling c.show(), it prints the correct results on the next line, but when i dont, it returns the wrong result, not sure why.
```python
def test_offset() -> None:
    c = gf.Component()
    c.add_polygon([(0, 0), (0, 10), (10, 10), (10, 0)], layer=(1, 0))
    c.offset(layer=(1, 0), distance=10)
    print(c.dbbox(get_layer((1, 0))))
```
```bash
> (0,0;10,10)
```

```python
def test_offset() -> None:
    c = gf.Component()
    c.add_polygon([(0, 0), (0, 10), (10, 10), (10, 0)], layer=(1, 0))
    c.offset(layer=(1, 0), distance=10)
    c.show()
    print(c.dbbox(get_layer((1, 0))))
```
```bash
> (-10,-10;20,20)
```

MatthewMckee4, 2025-01-14:
@joamatab 
@sebastian-goeldi

sebastian-goeldi, 2025-01-14:
Problem is the following logic:

```python
        layer_index = get_layer(layer)
        region = kdb.Region(self._kdb_cell.begin_shapes_rec(layer_index))
        region.size(+distance_dbu).size(-distance_dbu)
        self.remove_layers([layer])
        self._kdb_cell.shapes(layer_index).insert(region)
```

This potentially (or actually) writes to layer_index whoch doesn't exist anymore after removal. On write I assume it gets remapped to the old index, which is why it is there again.

The second layer_index must be retrieved with get_layer again

MatthewMckee4, 2025-01-14:
I can have a go at fixing it in a bit, unless you know how to do it rn

sebastian-goeldi, 2025-01-14:
`self._kdb_cell.shapes(layer_index).insert(region)` -> `self._kdb_cell.shapes(get_layer(layer).insert(region)`

At least I think. I am not 100% sure, but it's likely this logic

MatthewMckee4, 2025-01-15:
this isnt working unfortunately

sebastian-goeldi, 2025-01-15:
Hmmm, let me check a bit later today

sebastian-goeldi, 2025-01-16:
This is related to the iterator leaking in KLayout, fixed by https://github.com/KLayout/klayout/pull/1956

There is no release yet, so either we need to pin to 0.29.10 or we wait for 0.29.11

closed 2025-02-12 by sebastian-goeldi