gdsfactory/gdsfactory #3613: RuntimeError: Internal error: src/db/db/dbLayout.cc:1861 topological_sort () was not true in Cell.called_cells

joamatab, 2025-01-17:
```python
    import gdsfactory as gf

    c = gf.c.bend_euler_all_angle(angle=45)
    c.write_gds("bend_euler_all_angle.gds")
    c.show()
```

@MatthewMckee4 
@sebastian-goeldi

MatthewMckee4, 2025-01-17:
related to https://github.com/gdsfactory/kfactory/issues/550

MatthewMckee4, 2025-01-17:
i think it just fails in general (not sure though)

sebastian-goeldi, 2025-01-19:
Should be fixed now that klayout 0.29.11 released

joamatab, 2025-01-19:
it's not fixed, i upgraded to latest klayout and the issue still persists

MatthewMckee4, 2025-01-19:
yeah, still there

sebastian-goeldi, 2025-01-19:
Sorry, yeah, there's a bug in kfactory. It's trying to insert the cell into itself

sebastian-goeldi, 2025-01-25:
This should be fixed in >=0.23.2, yes?

joamatab, 2025-01-25:
yes, it was, we still has some issues with some custom clear_cache functions

i'll try to create a new issue for that

closed 2025-01-25 by joamatab

sebastian-goeldi, 2025-01-25:
oh, we probably should use klayout >=0.29.11, I don't think kfactory pinned it. But soon we will use `Cell.locked` to actually lock creation of shapes and modifications, so requiring 0.29.11 makes sense

joamatab, 2025-01-26:
even with klayout 0.29.11 i  still see that issue

sebastian-goeldi, 2025-01-26:
cannot reproduce

- recreated env
- installed latest dependencies
- ran MWE successfully
- check kfactory for relevant code which is there