gdsfactory/gdsfactory #3645: import gdsfactory as gf

Deepa0325, 2025-02-03:
import gdsfactory as gf

c = gf.Component()

ref1 = c.add_ref(gf.components.rectangle(size=(10, 10), layer=(1, 0)))

ref2 = c.add_ref(gf.components.text("Hello", size=10, layer=(2, 0)))

ref3 = c.add_ref(gf.components.text("world", size=10, layer=(2, 0)))

ref1.xmax = ref2.xmin - 5

ref3.xmin = ref2.xmax + 2

ref3.rotate(30)

c.show()

I have an error in this coding, which indicates the 'gdsfactory'- I can't rectify the error. Could you `help` me with this?

_Originally posted by @Deepa0325 in https://github.com/gdsfactory/gdsfactory/issues/3627#issuecomment-2629890285_

sebastian-goeldi, 2025-02-03:
If you don't post the error, it's hard to debug.

I tried locally with the last commit from `main`

```python
import gdsfactory as gf

c = gf.Component()

ref1 = c.add_ref(gf.components.rectangle(size=(10, 10), layer=(1, 0)))
ref2 = c.add_ref(gf.components.text("Hello", size=10, layer=(2, 0)))
ref3 = c.add_ref(gf.components.text("world", size=10, layer=(2, 0)))

ref1.xmax = ref2.xmin - 5
ref3.xmin = ref2.xmax + 2

ref3.rotate(30)

c.show()
```

This produces as expected the gds

[image]

closed 2025-02-04 by joamatab

Deepa0325, 2025-02-04:
`Cell` In[1], [line 1](vscode-notebook-cell:?execution_count=1&line=1)
----> [1](vscode-notebook-cell:?execution_count=1&line=1) import gdsfactory as gf
      [3](vscode-notebook-cell:?execution_count=1&line=3) c = gf.Component()
      [5](vscode-notebook-cell:?execution_count=1&line=5) ref1 = c.add_ref(gf.components.rectangle(size=(10, 10), layer=(1, 0)))

ModuleNotFoundError: No module named 'gdsfactory' 

This is the error I got.

sebastian-goeldi, 2025-02-04:
Yes, but this is a vscode or python problem. I am pretty sure you do not have the correct python env in vscode configured.

Please check the settings of vscode and that it uses the correct python executable.

Can you run the file directly with`python file.py`? Either in the vscode terminal or pure shell and activate env.