gdsfactory/gdsfactory #3654: Need help fixing this routing function

joamatab, 2025-02-10:
Need help fixing this routing function

```python

"""LVS demo."""

from __future__ import annotations

import gdsfactory as gf
from cspdk.si220 import cells

@gf.cell
def demo_astar_electrical() -> gf.Component:
    c = gf.Component()
    cross_section_name = "metal_routing"
    port_prefix = "e"
    bend = gf.components.wire_corner

    cross_section = gf.get_cross_section(cross_section_name)
    w = gf.components.straight(cross_section=cross_section)
    left = c << w
    right = c << w
    right.rotate(90)  # type: ignore[arg-type]
    right.move((168, 63))

    obstacle = gf.components.rectangle(size=(250, 3), layer="PAD")
    obstacle1 = c << obstacle
    obstacle2 = c << obstacle
    obstacle3 = c << obstacle
    obstacle4 = c << obstacle
    obstacle4.rotate(90)  # type: ignore[arg-type]
    obstacle1.ymin = 50
    obstacle1.xmin = -10
    obstacle2.xmin = 35
    obstacle3.ymin = 42
    obstacle3.xmin = 72.23  # type: ignore
    obstacle4.xmin = 200
    obstacle4.ymin = 55
    port1 = left.ports[f"{port_prefix}1"]
    port2 = right.ports[f"{port_prefix}2"]

    gf.routing.route_astar(
        component=c,
        port1=port1,
        port2=port2,
        cross_section=cross_section,
        resolution=10,
        distance=12,
        avoid_layers=("PAD",),
        bend=bend,
    )
    return c

if __name__ == "__main__":
    c = demo_astar_electrical()
    c.show()
```

sebastian-goeldi, 2025-02-12:
as far as I can tell, the astar rout can produce non-manhattan paths. I am not sure where this comes from or why, but the simplify in shapely doesn't make it better for sure

joamatab, 2025-02-19:
fixed in gdsfactory9

closed 2025-02-19 by joamatab