gdsfactory/gdsfactory #3658: Route bundle electrical fails when pads and routes are on layer (1, 0)

wagnojunior, 2025-02-11:
The following example from the docs words fine.

```python
c = gf.Component()
pt = c << gf.components.pad_array(port_orientation=270, columns=3, centered_ports=False)
pb = c << gf.components.pad_array(port_orientation=90, columns=3, centered_ports=False)
pt.move((100, 300))

routes = gf.routing.route_bundle_electrical(
    c,
    pb.ports,
    pt.ports,
    start_straight_length=30,
    separation=30,
    cross_section="metal_routing",
)
c.plot()
```

[image]

However, when I change the layer of the pads to (1, 0) and add the `layer` and `route_width` parameters the following error appears:
```python
c = gf.Component()
pt = c << gf.components.pad_array(port_orientation=270, columns=3, centered_ports=False, layer=(1, 0))
pb = c << gf.components.pad_array(port_orientation=90, columns=3, centered_ports=False, layer=(1, 0))
pt.move((100, 300))

routes = gf.routing.route_bundle_electrical(
    c,
    pb.ports,
    pt.ports,
    start_straight_length=30,
    separation=30,
    layer=(1, 0),
    route_width=15
)
c.plot()
```

**ValueError: Taper component should have two ports of port_type='electrical'! Got ['Port(name: o1, dwidth: 100.0, trans: r180 *1 0,0, layer: WG (1/0), port_type: optical)', 'Port(name: o2, dwidth: 15.0, trans: r0 *1 10,0, layer: WG (1/0), port_type: optical)**'].

tvt173, 2025-02-11:
hi @wagnojunior , it's because you have auto-tapering enabled and you are using the WG layer, which is optical. if you turn off auto-tapering and give the pads a bit more space, it works

```python
import gdsfactory as gf

c = gf.Component()
pt = c << gf.components.pad_array(port_orientation=270, columns=3, centered_ports=False, layer=(1, 0))
pb = c << gf.components.pad_array(port_orientation=90, columns=3, centered_ports=False, layer=(1, 0))
pt.move((200, 600))

routes = gf.routing.route_bundle_electrical(
    c,
    pb.ports,
    pt.ports,
    start_straight_length=30,
    separation=30,
    layer=(1, 0),
    route_width=15,
    auto_taper=False,
)
c.plot()
```

auto-tapering will automatically insert tapers to ports with mismatched widths. taper functions are registered by layer, in the `layer_transitions` dictionary of the PDK. so when you make this electrical device use the waveguide layer, it fetches those optical components for the tapers and complains. this is because you are using the default technology of the generic PDK. you could always create your own PDK with different settings and layer mappings

wagnojunior, 2025-02-12:
Amazing! Thank you for the tips~!