gdsfactory/gdsfactory #3662: `route_bundle` docstring references `cross_section.gap`, which does not exist.

wagnojunior, 2025-02-12:
- GDS Factory version: 8.32.1

The docstring of the `route_bundle` function states that the separation value defaults to `cross_section.width + cross_section.gap`.

```python
"""separation: bundle separation (center to center). Defaults to cross_section.width + cross_section.gap"""
```

However, as far as I know, a cross section does not have a `gap` parameter.

```python
"""Return CrossSection.

    Args:
        width: main Section width (um).
        offset: main Section center offset (um).
        layer: main section layer.
        sections: list of Sections(width, offset, layer, ports).
        port_names: for input and output ('o1', 'o2').
        port_types: for input and output: electrical, optical, vertical_te ...
        bbox_layers: list of layers bounding boxes to extrude.
        bbox_offsets: list of offset from bounding box edge.
        cladding_layers: list of layers to extrude.
        cladding_offsets: list of offset from main Section edge.
        cladding_simplify: Optional Tolerance value for the simplification algorithm. \
                All points that can be removed without changing the resulting. \
                polygon by more than the value listed here will be removed.
        cladding_centers: center offset for each cladding layer. Defaults to 0.
        radius: routing bend radius (um).
        radius_min: min acceptable bend radius.
        main_section_name: name of the main section. Defaults to _default
(...)
"""
```

wagnojunior, 2025-03-04:
I see. So does that mean that the documentation needs to be updated?

sebastian-goeldi, 2025-03-11:
Yes.

Sorry about the late respone