gdsfactory/gdsfactory #3675: Use `Sequence` instead of `tuple` or `list` for function args

95-martin-orion, 2025-02-14:
**Is your feature request related to a problem? Please describe.**
Annotating function arguments with `tuple` or `list` will cause typecheckers to expect values passed to those arguments to have those exact types. For example, this will cause a error in typechecking:

```py
def foo(args: list[str]) -> str:
    return args[0]

my_args = ("a", "b")
print(foo(my_args))  # error: my_args is a tuple, not a list
```

This has come up in several recent type-annotations changes, which - while more complete type annotations are greatly appreciated! - have caused a considerable amount of toil on my downstream project, as mismatched types that otherwise work with a function are now rejected by the typechecker.

**Describe the solution you'd like**
If an argument can be a tuple or a list (or a numpy array, etc.) without breaking the function behavior, use the `Sequence` type from `collections.abc` instead. This type will accept any sequence-like object:

```py
from collections.abc import Sequence

def foo(args: Sequence[str]) -> str:
    return args[0]

my_tuple = ("a", "b")
my_list = ["a", "b"]
my_ndarray = np.array(["a", "b"])

print(foo(my_tuple))  # OK
print(foo(my_list))  # OK
print(foo(my_ndarray))  # OK
```

- [x] I understand that, as an open-source project, the maintainers may have limited capacity to address every request. I am open to contributing by developing the feature myself or hiring someone to implement it (contact@gdsfactory.com). Otherwise, I understand that this feature may not be prioritized.

95-martin-orion, 2025-02-14:
The same logic applies for other `collections.abc` types, although `Sequence` is by far the most common culprit.

Note that this does NOT apply to function return types; a function knows the exact type(s) it is able to return, and callers benefit from knowing that exact type.

sebastian-goeldi, 2025-02-14:
Just one word of caution. Please don't apply this to the components / `@cell` decorated function. The component functions except a hashable Sequence/Mapping (or dict/list which as specially handled). So there it would break stuff.

Is that a reasonable compromise for now @95-martin-orion ?

95-martin-orion, 2025-02-14:
> Is that a reasonable compromise for now @95-martin-orion ?

Yes, if there's special handling necessary to keep the cache happy, that should take priority over this request.

sebastian-goeldi, 2025-02-14:
There are two things:

- The following can be very confusing, because it passes a type checker, but fails due to UserList not being hashable:
    ```python
    import gdsfactory as gf
    
    from collections import UserList
    from typing import Sequence
    
    
    @gf.cell
    def test_cell(name_list: Sequence[str]) -> gf.Component:
        c = gf.Component()
        return c
    
    
    class MyList(UserList[str]):
        pass
    
    
    c = gf.Component()
    c << test_cell(["abc", "de", "fg"])
    c << test_cell(MyList(("abc", "de", "fg")))
    
    c.show()
    ```
- If two things were hashable, but don't resolve to the same order while creating the same name, you can have a duplicate name problem which can be very hard to debug