gdsfactory/gdsfactory #3696: Documentation request: Who should use gdsfactory vs kfactory?

yoshi74ls181, 2025-03-02:
Now that kfactory has passed v1.0.0, I think it is a good time to clarify in README who should be using gdsfactory vs kfactory. Otherwise, a newcomer might assume that kfactory only exists as a backend for gdsfactory and has no use case on its own. Or that gdsfactory is only a compatibility layer on top of kfactory for people who has been using gdsfactory since when its backend was gdspy/gdstk.

- [x] I understand that, as an open-source project, the maintainers may have limited capacity to address every request. I am open to contributing by developing the feature myself or hiring someone to implement it (contact@gdsfactory.com). Otherwise, I understand that this feature may not be prioritized.

sebastian-goeldi, 2025-03-03:
Hi,

That really depends what your goal would be I guess. We can put something in the documentation, but i think it will be a pretty biased view no matter what. But in general I would say:

Choose gdsfactory if any of the following applies:
- Your base is gdsfactory already
- You don't need fine-grained access (only design in microns, no special access to dbu necessary)
- Use of pre-baked component functions and utilities of gdsfactory

Otherwise, you can use either. Only if you need full control over dimensions, kfactory is really more suitable. Or you have deeper requirements, such as strict separation of cells, in which case you would need full access to the different KCLayout/PDKs of kfactory. gdsfactory does not support that (yet).

tvt173, 2025-03-03:
Hi @yoshi74ls181, おはようございます。

yes, i agree it would be somewhat opinionated-- i'll add my opinion as well:

gdsfactory is generally designed to be more approachable... it is better documented. units are in microns, rather than dbu. it is more compatible with many more plugins, and also our paid offerings, such as [gdsfactory+](https://doplaydo.github.io/doCode/). 

kfactory is a bit more raw, a bit closer to the klayout API. it uses dbu directly by default. it can support some low-level features, as sebastian mentions, that we have not bothered to include in gdsfactory yet.

so my personal opinion would be to use gdsfactory, unless you have a reason to use kfactory directly (such as the reasons sebastian points out). most people and organizations use gdsfactory, so it is a bit better supported, though there are a few notable ones using kfactory directly as well.

i don't think it would hurt to add a summary of these different viewpoints to the readme of kfactory though, as you suggest. chatgpt helped me with such a statement:

## Choosing Between GDSFactory and KFactory
KFactory is a powerful tool for working with GDSII layouts, but choosing between KFactory and GDSFactory depends on your specific needs. Below is a general guideline to help you decide:

### Use GDSFactory if:

- You are already working within the GDSFactory ecosystem.
- You prefer a more approachable, well-documented framework with support for many plugins and integrations.
- You design primarily in microns and do not require fine-grained control over database units (dbu).
- You want access to pre-built component functions and utilities for streamlined layout generation.

### Use KFactory if:

- You need precise, low-level control over layout dimensions, including direct manipulation of dbu.
- You require strict separation of cells or full access to multiple KCLayout/PDKs.
- You prefer working closer to the KLayout API for maximum flexibility.
- You have specialized requirements that GDSFactory does not yet support.

While most users and organizations opt for GDSFactory due to its broader support and ease of use, KFactory remains a strong choice for those who need deeper control over their designs. Both tools serve different but complementary purposes, and the best choice depends on your workflow and requirements.

yoshi74ls181, 2025-03-03:
@sebastian-goeldi @tvt173 Thank you very much for the clarification. Could you also comment on who the main contributers are for kfactory vs gdsfactory? By biased, do you mean that I would get a different answer from a kfactory contributor?

tvt173, 2025-03-03:
Hi @yoshi74ls181 , I'll give you a brief history: gdsfactory came first. Then Sebastian created kfactory to address some shortcomings of gdsfactory. Particularly dbu snapping issues, but also because klayout is more reliable in general than gdspy/gdstk. Then after some time we decided we wanted to move gdsfactory also to use kfactory as a backend and effectively merge the two projects. So that's where we are now. Sebastian still mostly uses kfactory directly. Most of the rest of us use gdsfactory, with the kfactory backend, and that's generally what we recommend to others also

sebastian-goeldi, 2025-03-03:
Yes, as Troy pointed out, that's the history behind it.

W.r.t. contributors, you can see for yourself, here https://github.com/gdsfactory/gdsfactory/graphs/contributors and here https://github.com/gdsfactory/kfactory/graphs/contributors

You will see a large overlap ;). I have focused mostly on kfactory or kfactory integration/adoption in gdsfactory, Joaquin on some back compatibility of kfactory for gdsfactory, Troy and Floris on simulations and netlists, and Matthew has helped a ton of making kfactory more friendly towards gdsfactory and also making gdsfactory type checker compliant. Others I am less sure, but a lot helped with cross sections, simulations, and/or components