gdsfactory/gdsfactory #3697: How to adapt code for the latest version of GDSFactory

ZhimingShi, 2025-03-02:
Hi, I'm currently using GDSFactory v8.21. After updating to the latest v9.02, many of my existing scripts are throwing errors due to incompatibility. Besides manually fixing each issue based on error messages, is there a way to automate the adaptation to the new version? GDSFactory updates very frequently, and modifying code after every release is a massive task.

MatthewMckee4, 2025-03-02:
Can you show some of the errors you are seeing?

sebastian-goeldi, 2025-03-03:
Hi @ZhimingShi

No there are currently no migration scripts, as we didn't break anything intentionally, so more or less everything except the constructor for ports should be considered a bug or at least worth an issue.

Could you provide example(s) of what broke?

joamatab, 2025-03-03:
You always have the option to stick with your current version if upgrading isn't ideal for your workflow. That said, we provide [migration guidelines](https://gdsfactory.github.io/gdsfactory/migration.html) in the docs to help with the transition.

sebastian-goeldi, 2025-03-03:
Is the guide for 8 to 9 still accurate? I thought @MatthewMckee4 made the types in typings available again?

MatthewMckee4, 2025-03-03:
Yeah we could change that, i had only written that before i changed the typings back, ill make a PR