gdsfactory/gdsfactory #3721: Component translation after rotation has unexpected offset

octavephotonics, 2025-03-12:
**Describe the bug**
What's the bug?
Applying a rotation transformation to certain components leads to incorrect position values for the component. The rotation works correctly, but subsequent attempts to position the rotated component via xmin, ymin, etc do not produce the expected behavior. I also have tested this on a Path object and found similar behavior.
 
Here is an image output from the example below that should create two triangles (one of which is rotated) with the lower left corner of each at the same location.
[image]

**To Reproduce**
Here is a minimal working example that creates a triangle, and adds two references to a component, one of which gets rotated. The xmin and ymin commands are used to attempt to position the lower left corner of each triangle at the same location (0,0).
```
import gdsfactory as gf

c = gf.Component()

fc = gf.Component('triangle')
p1 = (0,0); p2 = (6, 0); p3 = (3,3)
points = [p1, p2, p3]
p = fc.add_polygon(points, layer=(0,0))

#add a non-transformed reference
pref = c << fc
pref.xmin = 0
pref.ymin = 0

#transform a new reference and try to align lower left corner with pref
pref2 = c << fc
pref2.rotate(30)
pref2.ymin = 0
pref2.xmin = 0

c.show()
```

**Expected behavior**
Setting position variables on rotated objects should account for any previously applied rotation transformation.

**Environment (please complete the following information):**

- [X] I have reviewed the documentation, discussions and issues and found no relevant solution.
- [X] I understand that this is an open-source project, and maintainers may not have the resources to address every issue. I am prepared to contribute by fixing the issue myself or hiring someone to do so, if needed, and accept that the issue may not be resolved otherwise.
- [X] I am using the latest version of GDSFactory, with Python 3.11, 3.12, or 3.13.

Relevant software versions:
Python 3.12.7
gdsfactory 9.1.0
kfactory 1.2.3

octavephotonics, 2025-03-13:
Ok that makes sense if the position values are just based on the transformed bbox. Still, this changes previous behavior breaking old scripts, and is also less intuitive IMO. I agree though that in the toy triangle example it's easy to just change the order of operations to get the desired behavior, but this is not always true for more complicated structures. 

Generally the user cares about the shapes in their structure and need to make position changes based on those, regardless of the orientation of the original bbox, which would now have to be a separate geometry to keep track of, right?

Running the same code above with, for example, gdsfactory 7.27.0 produces the expected result:
[image]

sebastian-goeldi, 2025-03-14:
I understand, but there is not much we can do about that. All the other operations would be very expensive. In klayout there is no way to calculate the gf7/gdstk style bounding box in klayout. So, there is nothing we can do about that.

Timemakesmegrow, 2025-03-17:
hahaha,I've also complained about this problem.

octavephotonics, 2025-03-18:
Thanks for the comments and suggestions! My main issue with the proposed solutions is that they make the user keep track of extra geometries or do operations in a specific order, which in general may not be the most intuitive way to build geometries when things get more complicated. To make things easier, at the expense of extra computation, I think I'm looking for a (optional) way to recompute the bbox. In the meantime, I've been able to achieve this for different geometries by either creating a copy of the component or calling flatten() depending on the context. In the simple triangle example, calling copy before rotating seems to do the trick.

```
import gdsfactory as gf

c = gf.Component()

fc = gf.Component('triangle')
p1 = (0,0); p2 = (6, 0); p3 = (3,3)
points = [p1, p2, p3]
p = fc.add_polygon(points, layer=(0,0))

#add a non-transformed reference
pref = c << fc
pref.xmin = 0
pref.ymin = 0

#transform a new reference and try to align lower left corner with pref
fcrot = fc.copy()
fcrot.rotate(30)
#fcrot.flatten()

pref2 = c << fcrot
pref2.ymin = 0
pref2.xmin = 0

c.show()
```

[image]

[image]

sebastian-goeldi, 2025-03-18:
I can give you the code to do that, nonetheless I have to say:

The whole premise is a bit weird to me. If you rotate your triangle like this (i.e. none-90°angles), none of the point of the triangle with the sole exception of the point in the origin are on grid. So, if you want to be entirely accurate with your translation, it's impossible (because the cell itself cannot be moved to any position but something on-grid).

In order to solve this, use a boolean structure, i.e. a 

E.g. for the triangle you can do:

`bbox = gf.kdb.Region(triangle_cell.begin_shapes_rec(LAYER.I_CARE_ABOUT).transform(triangle_instance.cplx_trans).bbox().to_dtype(triangle_cell.kcl.dbu)`

You could do this for any number of layers, but this gets expensive very fast for complex cells.

This complexity is also why I don't really see the point of exposing this, as the accuracy is "more or less" at best anyways.

joamatab, 2026-07-21:
Correction to the shell-mangled identifiers in my prior comment: DInstance.xmin/ymin use the transformed cell bbox. Exact per-layer bounds can be computed with kdb.Region(cell.begin_shapes_rec(layer_index)).transform(instance.cplx_trans).bbox(), with the performance and DBU-grid caveats described above.