gdsfactory/gdsfactory #3724: Methods to improve the speed of operations when a large number of graphic logic operations are involved

Timemakesmegrow, 2025-03-15:
Hello, unlike chips, I specialize in metasurfaces. In the process I use it often requires a lot of data to be imported and calculated.For various strange pillers on the metasurface, I often need to perform logical operations between two graphics and obtain 2D graphics.Of course, it is feasible to calculate the coordinates of each point and bring them into polygon, but it is too much for me to do the job.Of course, there are some graphics in gdsfactory, which are very convenient to call, but if it is a graph without it, I prefer logical operations between graphics.However, when it comes to a large amount of data such as a 5000*5000 matrix, logical calculations are performed one element by one, and after I check, the calculation speed is very slow.

So I'm wondering if you have a recommended toolkit to handle similar situations, speed up the run, or is it possible to reduce the time of logical operations? In this way, the logical operations between graphics may be a bit useless

- [yes ] I understand that, as an open-source project, the maintainers may have limited capacity to address every request. I am open to contributing by developing the feature myself or hiring someone to implement it (contact@gdsfactory.com). Otherwise, I understand that this feature may not be prioritized.

sebastian-goeldi, 2025-03-15:
Not sure I fully follow what you are trying to do.

Are you trying to perform booleans between different layers in a gds or.is the data external and you want to efficiently bring it into gdsfactory. Or are the booleans too slow?

If it's about booleans you should look into using regions probably. E.g. the outline here uses regions https://gdsfactory.github.io/gdsfactory/notebooks/04_components_geometry.html#outline

If regions are too slow for your purposes, you can use the tilingprocessor of klayout, but then I would suggest to loom at the klayout forum for inspiration (setup etc can be quite complex and heavily depend on exact usecase)

Timemakesmegrow, 2025-03-17:
Thank you for your reply.For example, there are two shapes, one is a square and the other is a circle.The length of a square is a parameter, and the radius of a circle is also a parameter. I need to superimpose the square and circle in the center and then subtract the circle with the square to get the remaining shape.So here is bool operation. When I change the parameter value, it means that every time I change the graph, I have to go through bool operation. After I checked the bool operation, it consumes a lot of memory.
But my task requires about 5000*5000 operations, so bool operations are convenient but not the best solution for my task. So I want to ask if there is a better way

sebastian-goeldi, 2025-03-17:
Without an example it's really hard to help. It all depends on the size etc. If you are talking about 5000 x 5000 shapes, yeah, no matter what you will do, your performance will probably be bad. If it's a repeating pattern, you can use arrays of references.