gdsfactory/gdsfactory #3744: gf.get_cell doesn't pass settings

Alessio-Zicoschi, 2025-03-25:
**Describe the bug**
Maybe I am missing something, but it seems to me that the get_cell function returns a cell without the extra kwargs I pass. On the other hand, get_component gets me the component with the additional kwargs, as I would expect. Basically, I was hoping I could use get_cell instead of functors partial for cells in my pdk. When building complicated components this feature could be helpful.

**To Reproduce**
```
import gdsfactory as gf

print(gf.get_active_pdk().name)

call = gf.get_cell("straight", length=20)
c1 = gf.get_component(call)
c1.show()
print(c1.info.length)

c2 = gf.get_component("straight", length=20)
c2.show()
print(c2.info.length)
```
**Expected behavior**
I am working in the generic pdk, gdsfactory version 9.3.0. I would expect these two to have the same output, while I get a length of 10 (default value for straight) inn the first case and 20 in the second one.

**Suggested fix**
I guess the get_cell function should return something like partial(cell, **kwargs), which seems like it is implemented in the source code. Maybe the kwargs are lost somewhere.

**Environment (please complete the following information):**

- [ x] I have reviewed the documentation, discussions and issues and found no relevant solution.
- [ x] I understand that this is an open-source project, and maintainers may not have the resources to address every issue. I am prepared to contribute by fixing the issue myself or hiring someone to do so, if needed, and accept that the issue may not be resolved otherwise.
- [ x] I am using the latest version of GDSFactory, with Python 3.11, 3.12, or 3.13.

Please provide the output for the following code:

```python
3.12.9 | packaged by conda-forge | (main, Mar  4 2025, 22:44:42) [Clang 18.1.8 ]
/Users/alessiozicoschi/miniconda3/envs/omgdslib_env/bin/python
                                                                               Modules                                                                               
┏━━━━━━━━━━━━━━━━┳━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━┳━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━┓
┃        Package ┃ version                                                                ┃                                                                    Path ┃
┡━━━━━━━━━━━━━━━━╇━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━╇━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━┩
│         python │ 3.12.9 | packaged by conda-forge | (main, Mar  4 2025, 22:44:42)       │          /Users/alessiozicoschi/miniconda3/envs/omgdslib_env/bin/python │
│                │ [Clang 18.1.8 ]                                                        │                                                                         │
│     gdsfactory │ 9.3.0                                                                  │ /Users/alessiozicoschi/miniconda3/envs/omgdslib_env/lib/python3.12/sit… │
│         devsim │ not installed                                                          │                                                                         │
│        femwell │ not installed                                                          │                                                                         │
│ gdsfactoryplus │ not installed                                                          │                                                                         │
│       gplugins │ not installed                                                          │                                                                         │
│       kfactory │ 1.2.5                                                                  │ ['/Users/alessiozicoschi/miniconda3/envs/omgdslib_env/lib/python3.12/s… │
│         lumapi │ not installed                                                          │                                                                         │
│           meep │ not installed                                                          │                                                                         │
│           meow │ not installed                                                          │                                                                         │
│            ray │ not installed                                                          │                                                                         │
│            sax │ not installed                                                          │                                                                         │
│         tidy3d │ not installed                                                          │                                                                         │
│          vlsir │ not installed                                                          │                                                                         
```

Alessio-Zicoschi, 2025-03-27:
Hi,
I see, thanks for your answer! I understand then that this is not a bug. 
I believe it could be useful if get_cell could accept also kwargs for the cell, encoding them in a callable for later use. I am currently using the following workaround:
```
import gdsfactory as gf
from functools import partial

print(gf.get_active_pdk().name)

current_get_cell = gf.get_cell("straight", length=20)
c1 = gf.get_component(current_get_cell)
print(c1.info.length)

c2 = gf.get_component("straight", length=20)
print(c2.info.length)

desired_get_cell = partial(gf.get_cell("straight"), length=20)
c1 = gf.get_component(desired_get_cell)
print(c1.info.length)
```

`desired_get_cell` already encodes the length information in the callable. 
I find this advantageous when building components with few levels of hierarchy and if I need to sweep some parameters at different levels. It avoids me generating cells for all lengths like `straight_10`, `straight_11` and so on and having to store them in my pdk. I can, on the other hand, create these callables at run time, put them in a list and use each one with different arguments, which might depend on other parameters of the components I am building. I hope this is clear.
Thank you for your availability!

joamatab, 2025-03-27:
You could use partials for that

```python
straight10=partial(gf.c.straight, length=10)
straight20=partial(gf.c.straight, length=20)
```

I'll close the issue for now

closed 2025-03-27 by joamatab