gdsfactory/gdsfactory #3745: How to route AWG waveguide

ZhimingShi, 2025-03-26:
Hello, I'm using GDSFactory to design an AWG (Arrayed Waveguide Grating) and have encountered two issues.

1. The first issue is with the FPR (Free Propagation Region) connection ports. As shown in the diagram below, the port orientations align radially with the center of their respective arcs (i.e., their angles are not 0°, 90°, 180°, or 270°). How can I properly connect these two FPRs?

[image]
FPR design

[image]
FPR one of the ports, not 0, 90, 180, 270°

2. The second issue is that the AWG design requires a fixed length difference L，between adjacent waveguides: the second waveguide must be longer than the first by L， the third longer than the second by L，and so on. How can I design the routing to achieve this incremental path length difference?

[image]

[image]

I want to implement a connection similar to these two above

joamatab, 2025-03-26:
Hi,

Where can we find the code for what you've been working on?

I had started something similar a while back [here](https://github.com/gdsfactory/gdsfactory/tree/add_awg):

It still needs more work to be fully functional, but it could be a good starting point.

tvt173, 2025-03-27:
hi @ZhimingShi ,

you will need to two things primarily to make this work:
1. Convert the `free_propagation_region` to a [virtual cell](https://gdsfactory.github.io/gdsfactory/notebooks/12_grid_snapping.html#solution), so that it can have off-grid and off-angle ports
2. Connect the two FPRs by [all-angle routes ](https://gdsfactory.github.io/gdsfactory/notebooks/04_routing.html#route-bundle-all-angle). If your FPR is curved, you may want to inset your ports by a small amount so there are no gaps at the edges.

that will at least get you to a properly-connected AWG. to achieve the desired dL value between adjacent routes, I think you may need to do a little more manual work (I don't think we have a good automatic way to do this yet for all-angle routes, right @sebastian-goeldi ?).

Perhaps the easiest way to do this would be to create some intermediate straights at the center of the route, as shown below... Space them out by the default minimum spacing to start, try routing, figure out the base dL between the arms, then adjust the spacing of the straights as necessary to compensate and re-create the routes.

[image]

if you submit a PR with your design, we'd be happy to help review it and add it to the generic PDK!

sebastian-goeldi, 2025-03-28:
No, the all-angle router cannot directly handle this. But you can kind of achieve it by using it to route individual ports and then supplying the dL.

AndreaWCKim, 2025-06-11:
I am doing a similar work. But I am new for gdsfactory. 
The image shown in below.. includes
1. exponential taper
2. angled free propagaion region
3. gradually seperated tapers for FPR to array transtion
4. straight waveguides
5. arc waveguides

The path difference of each array is fixed value, but half of dL. 
I just make half.

[image]

PongpopSukanunta, 2025-11-11:
Hi, I am also working on the AWG, but I am new to GDSFactory and I want some opinion on my method. I calculated boundaries of both the FPR and all of the arrayed waveguides first, then  I use add_polygons() to create the layout. Would this be something you think is appropriate for the AWG layout?