gdsfactory/gdsfactory #3770: Excluding Meta Info by Default in gdsfactory for KLayout Previews to Resolve DRC Compatibility Issues

ZhimingShi, 2025-04-10:
Hello, I am using v8.19.0 gdsfactory for layout design. How to configure gdsfactory so that when running c.show(), the GDS graphics displayed in KLayout do not include Meta Info by default, rather than setting the saved file to exclude Meta Info during saving? The presence of Meta Info can cause compatibility issues during DRC checks.

[image]
I want to get rid of this message in the graphic in klayout after running c.show()

joamatab, 2025-04-10:
Hi, you can find the solution on the tutorial

https://gdsfactory.github.io/gdsfactory/notebooks/00_geometry.html#write

closed 2025-04-10 by joamatab

sebastian-goeldi, 2025-04-10:
You can replace the show function, but I am not sure whether that is available in 8.19 already.

```python
import gdsfactory as gf
from functools import partial
from pathlib import Path
from typing import Any

opts = gf.kf.save_layout_options()
opts.write_context_info = False

def show(
    layout: gf.kf.KCLayout | gf.kf.ProtoTKCell[Any] | Path | str,
    lyrdb: gf.kf.rdb.ReportDatabase | Path | str | None = None,
    l2n: gf.kf.kdb.LayoutToNetlist | Path | str | None = None,
    keep_position: bool = True,
    save_options: gf.kf.kdb.SaveLayoutOptions | None = None,
    use_libraries: bool = True,
    library_save_options: gf.kf.kdb.SaveLayoutOptions | None = None,
) -> None:
    save_options = save_options or gf.kf.save_layout_options()
    save_options.write_context_info = False
    return gf.kf.show(
        layout=layout,
        lyrdb=lyrdb,
        save_options=save_options,
        use_libraries=True,
        library_save_options=library_save_options,
    )

gf.kf.config.show_function = show

@gf.cell
def test(a: int) -> gf.Component:
    return gf.Component()

test(5).show()
```