gdsfactory/gdsfactory #3771: KLayout is in the batch mode after installing gdsfactory

cheng1214, 2025-04-10:
After I install gdsfactory, I cannot use SiEPIC tools from UBC. It raises the following error message:

_Exception: SiEPIC.utils.get_technology: this function can only be used in KLayout Application graphical interface mode.
  C:\Users\Rui\KLayout\salt\siepic_tools\python\SiEPIC\utils\__init__.py:362
  C:\Users\Rui\KLayout\salt\siepic_tools\python\SiEPIC\utils\__init__.py:732
  C:\Users\Rui\KLayout\salt\siepic_tools\python\SiEPIC\scripts.py:656
  C:/Users/Rui/KLayout/salt/siepic_tools/pymacros\Keybindings\Waveguides\Path to Waveguide.lym:5_

To find the reason, I checked the variable "_globals.Python_Env" and found it was "'KLayout_batch'".  But I supposed that variable should be "KLayout_GUI" in default?

I wondered if that problem could be related to any setting excuted by gdsfactory? How to solve this issue?

System Context：
Installed package: gdsfactory v8.17.0
Affected toolkit: SiEPIC Tools: 0.5.24; SiEPIC PDK: 0.4.34
Platform: KLayout 0.29.8 (Windows)

Thanking you!

cheng1214, 2025-04-11:
@thomaslima  Any suggestion or possible solution would be greatly appreciated!

thomaslima, 2025-04-11:
Are you running gdsfactory from klayout? This error seems to be coming from SiEPIC.utils.get_technology, so that's where you need to look for what condition it is checking to trigger this error.

joamatab, 2025-04-11:
GDSFactory not intended to use the python inside klayout

You can see the installation instructions in https://github.com/gdsfactory/ubc

closed 2025-04-17 by cheng1214