gdsfactory/gdsfactory #3783: Unable to import gdsfactory

darenc2, 2025-04-21:
Hi, I'm trying to get started and can't even import gdsfactory. I know I'm doing something wrong, but I just don't know what lol

Please provide the output for the following code:

```python
import sys
print(sys.version)
print(sys.executable)

import gdsfactory as gf
gf.config.print_version_plugins()
```

Output:
>print(sys.version)
3.12.7 | packaged by Anaconda, Inc. | (main, Oct  4 2024, 13:17:27) [MSC v.1929 64 bit (AMD64)]
>print(sys.executable)
C:\Users\daren\anaconda3\python.exe
>import gdsfactory as gf
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "C:\Users\daren\anaconda3\Lib\site-packages\gdsfactory\__init__.py", line 13, in <module>
    import kfactory as kf
  File "C:\Users\daren\anaconda3\Lib\site-packages\kfactory\__init__.py", line 32, in <module>
    from .layout import Constants, KCLayout, cell, vcell, kcl, kcls
  File "C:\Users\daren\anaconda3\Lib\site-packages\kfactory\layout.py", line 100, in <module>
    class KCLayout(
  File "C:\Users\daren\anaconda3\Lib\site-packages\pydantic\_internal\_model_construction.py", line 205, in __new__
    complete_model_class(
  File "C:\Users\daren\anaconda3\Lib\site-packages\pydantic\_internal\_model_construction.py", line 534, in complete_model_class
    schema = cls.__get_pydantic_core_schema__(cls, handler)
             ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "C:\Users\daren\anaconda3\Lib\site-packages\pydantic\main.py", line 643, in __get_pydantic_core_schema__
    return handler(source)
           ^^^^^^^^^^^^^^^
  File "C:\Users\daren\anaconda3\Lib\site-packages\pydantic\_internal\_schema_generation_shared.py", line 83, in __call__
    schema = self._handler(source_type)
             ^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "C:\Users\daren\anaconda3\Lib\site-packages\pydantic\_internal\_generate_schema.py", line 512, in generate_schema
    schema = self._generate_schema_inner(obj)
             ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "C:\Users\daren\anaconda3\Lib\site-packages\pydantic\_internal\_generate_schema.py", line 784, in _generate_schema_inner
    return self._model_schema(obj)
           ^^^^^^^^^^^^^^^^^^^^^^^
  File "C:\Users\daren\anaconda3\Lib\site-packages\pydantic\_internal\_generate_schema.py", line 591, in _model_schema
    {k: self._generate_md_field_schema(k, v, decorators) for k, v in fields.items()},
        ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "C:\Users\daren\anaconda3\Lib\site-packages\pydantic\_internal\_generate_schema.py", line 947, in _generate_md_field_schema
    common_field = self._common_field_schema(name, field_info, decorators)
                   ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "C:\Users\daren\anaconda3\Lib\site-packages\pydantic\_internal\_generate_schema.py", line 1134, in _common_field_schema
    schema = self._apply_annotations(
             ^^^^^^^^^^^^^^^^^^^^^^^^
  File "C:\Users\daren\anaconda3\Lib\site-packages\pydantic\_internal\_generate_schema.py", line 1890, in _apply_annotations
    schema = get_inner_schema(source_type)
             ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "C:\Users\daren\anaconda3\Lib\site-packages\pydantic\_internal\_schema_generation_shared.py", line 83, in __call__
    schema = self._handler(source_type)
             ^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "C:\Users\daren\anaconda3\Lib\site-packages\pydantic\_internal\_generate_schema.py", line 1871, in inner_handler
    schema = self._generate_schema_inner(obj)
             ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "C:\Users\daren\anaconda3\Lib\site-packages\pydantic\_internal\_generate_schema.py", line 789, in _generate_schema_inner
    return self.match_type(obj)
           ^^^^^^^^^^^^^^^^^^^^
  File "C:\Users\daren\anaconda3\Lib\site-packages\pydantic\_internal\_generate_schema.py", line 871, in match_type
    return self._match_generic_type(obj, origin)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "C:\Users\daren\anaconda3\Lib\site-packages\pydantic\_internal\_generate_schema.py", line 895, in _match_generic_type
    return self._union_schema(obj)
           ^^^^^^^^^^^^^^^^^^^^^^^
  File "C:\Users\daren\anaconda3\Lib\site-packages\pydantic\_internal\_generate_schema.py", line 1207, in _union_schema
    choices.append(self.generate_schema(arg))
                   ^^^^^^^^^^^^^^^^^^^^^^^^^
  File "C:\Users\daren\anaconda3\Lib\site-packages\pydantic\_internal\_generate_schema.py", line 512, in generate_schema
    schema = self._generate_schema_inner(obj)
             ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "C:\Users\daren\anaconda3\Lib\site-packages\pydantic\_internal\_generate_schema.py", line 789, in _generate_schema_inner
    return self.match_type(obj)
           ^^^^^^^^^^^^^^^^^^^^
  File "C:\Users\daren\anaconda3\Lib\site-packages\pydantic\_internal\_generate_schema.py", line 864, in match_type
    res = self._get_prepare_pydantic_annotations_for_known_type(obj, ())
          ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "C:\Users\daren\anaconda3\Lib\site-packages\pydantic\_internal\_generate_schema.py", line 1843, in _get_prepare_pydantic_annotations_for_known_type
    res = gen(obj, annotations, self._config_wrapper.config_dict)
          ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "C:\Users\daren\anaconda3\Lib\site-packages\pydantic\_internal\_std_types_schema.py", line 244, in path_schema_prepare_pydantic_annotations
    lax_schema=core_schema.union_schema(
               ^^^^^^^^^^^^^^^^^^^^^^^^^
TypeError: union_schema() got an unexpected keyword argument 'strict'

Any help would be much appreciated!

joamatab, 2025-04-22:
Hi,

Did you follow the installation [instructions](https://gdsfactory.github.io/gdsfactory/developer.html)?

I improved the instructions. Let me know if they work for you

## Installation for users in a new environment

We recommend `uv` for installing GDSFactory:

```bash
# On macOS and Linux.
curl -LsSf https://astral.sh/uv/install.sh | sh
```

```bash
# On Windows.
powershell -ExecutionPolicy ByPass -c "irm https://astral.sh/uv/install.ps1 | iex"
```

Then you can install gdsfactory with:

```bash
uv venv --python 3.12
uv pip install gdsfactory
```

darenc2, 2025-04-22:
Hi,
thanks so much for the quick response! I thought I followed the installation instructions, evidently not carefully enough. I reinstalled with uv and it worked! The only issue I ran into was that I had to downgrade to numpy 1.26.4 but everything seems to work for now.
Thanks,
Daren

darenc2, 2025-04-24:
I'm also using python 3.12.7, not sure why it told me to downgrade.

sebastian-goeldi, 2025-05-01:
Are you using a conda env or pure uv? That might play a role

darenc2, 2025-05-08:
Oh that's a good point, I am using conda env. I guess I wouldn't have this problem on pure uv?

closed 2026-02-07 by sebastian-goeldi