gdsfactory/gdsfactory #3811: write_gds with default save_metadata=True leads to unexpected behaviors in other softwares

pyMatJ, 2025-05-07:
Hi, 

This regards the `write_gds` option `save_metadata=True`. 
I use gdsfactory to generate gds files that are then processed through Beamer (from GenISys) for a 100kV EBL machine. 
I recently upgraded from v7 to v9 and noticed a change of behavior when importing my gds in beamer. GDS files produced by v9 are imported with warnings, as the top cell cannot be found by the software. This results in the layout showing the main layout but also all component definitions on top of each other at their "original" location (usually (0,0)). 
Trying other (proprietary) gds softwares (L-edit and Layout editor), similar warnings occur, with more or less the same behavior. 
Of course, everything is fine in klayout, and solvable in beamer by extracting only the cells of interest. 

A minimal example would be:
gdsfactory v7
```
c_dummy = gf.Component('dummy')
r = gf.components.rectangle(size=(1,10), layer=(1,0))
r = c_dummy.add_array(r, columns=10, rows=10, spacing=(2,12)) ## used in gdsfactory v7
r.move((-10,10))
c_dummy.write_gds('dummy_v7.gds')
```
gdsfactory v9
```
c_dummy = gf.Component('dummy')
r = gf.components.rectangle(size=(1,10), layer=(1,0))
r = c_dummy.add_ref(r, columns=10, rows=10, column_pitch=2, row_pitch=12) ## used in gdsfactory v9
r.move((-10,10))
c_dummy.write_gds('dummy_v9.gds')
c_dummy.write_gds('dummy_v9_nometadata.gds', with_metadata=False)
```

Producing the following results when imported in e.g. beamer 

| v7 or v9 without metadata: |  v9 with metadata  |
| --- | ---| 
| [image]  | [image] | 

You can also see that the import with metadata shows a layer (0,0) that is empty (no idea where it is coming from). 

I understand that this is easily fixed using the option `save_metadata=False` (or flattening before export but this is bad...). 
However it was unexpected, took me a while to pinpoint on a real design with more complex structures*, and there is no mention of this change of behavior in the migration guidelines (https://gdsfactory.github.io/gdsfactory/migration.html). I also don't fully understand what are these metadata, and how they are used. Maybe this is very klayout specific? Am I leaving out important stuff? Or is my workflow flawed somehow and this is normal and/or expected? 

I understand this is a collaborative project and this is probably really not urgent or even a bug. I would still suggest if you can consider:

- Referencing and/or documenting this in the migration guidelines and the documentation (and API doc)
- If this is really klayout-specific, maybe the default option should be changed to `False` to allow a wider compatibility?

I am more than happy to provide more information, and will also reach out to GenISys with this to see if they could make their import a bit more robust to that kind of things. 

Thank you for your time (and thank you so much for this really amazing work!!), and I am more than happy to add any relevant additional information if needed. 

*actually it was a gigantic array of a small rectangle, so it turned out I had one single instance of this tiny array off-grid and took a while to even see it, but it messed up a lot of things!

sebastian-goeldi, 2025-05-09:
This is intended behavior. I though we have the `write_gds(..., include_metadata=False)` documented somewhere.

I also intend to add a `write_fab(...)` at some point in order to facilitate this by default. But yes you are correct that this makes it incompatible with non-klayout flows under some circumstances.

On the other hand if we don't store this information, we lose tons of capabilities (e.g. reading ports from the gds, which is now possible among other information such as `settings` and `info`. Therefore, I don't consider disabling it by default as the way we would want to go. Documenting it better is probably advised though.

P.S. If you get such a gds which is incompatible, open it in klayout and disable this option while saving, and all metadata is gone, too.

[image]

joamatab, 2025-05-10:
Hi Matt,

you can find in the [docs](https://gdsfactory.github.io/gdsfactory/notebooks/00_geometry.html#write) how the metadata works

I also added it in the migration guidelines

closed 2025-05-10 by joamatab