gdsfactory/gdsfactory #3814: pydantic.errors.PydanticUserError: `CrossSectionModel` is not fully defined; you should define `KCLayout`, then call `CrossSectionModel.model_rebuild()`.

mithro, 2025-05-08:
When trying to import gdsfactory I get an error from pydantic.

```
tim@x1c-work:~/github/wafer-space/gf180mcu-test$ make shell
Starting shell with activated Python virtual environment...
(.venv) tim@x1c-work:~/github/wafer-space/gf180mcu-test$ python
Python 3.12.3 (main, Feb  4 2025, 14:48:35) [GCC 13.3.0] on linux
Type "help", "copyright", "credits" or "license" for more information.
>>> import gdsfactory
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "/home/tim/github/wafer-space/gf180mcu-test/.venv/lib/python3.12/site-packages/gdsfactory/__init__.py", line 13, in <module>
    import kfactory as kf
  File "/home/tim/github/wafer-space/gf180mcu-test/.venv/lib/python3.12/site-packages/kfactory/__init__.py", line 32, in <module>
    from .layout import KCLayout, cell, vcell, kcl, Constants
  File "/home/tim/github/wafer-space/gf180mcu-test/.venv/lib/python3.12/site-packages/kfactory/layout.py", line 1693, in <module>
    kcl = KCLayout("DEFAULT")
          ^^^^^^^^^^^^^^^^^^^
  File "/home/tim/github/wafer-space/gf180mcu-test/.venv/lib/python3.12/site-packages/kfactory/layout.py", line 232, in __init__
    cross_sections=CrossSectionModel(kcl=self),
                   ^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/tim/github/wafer-space/gf180mcu-test/.venv/lib/python3.12/site-packages/pydantic/main.py", line 253, in __init__
    validated_self = self.__pydantic_validator__.validate_python(data, self_instance=self)
                     ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/tim/github/wafer-space/gf180mcu-test/.venv/lib/python3.12/site-packages/pydantic/_internal/_mock_val_ser.py", line 100, in __getattr__
    raise PydanticUserError(self._error_message, code=self._code)
pydantic.errors.PydanticUserError: `CrossSectionModel` is not fully defined; you should define `KCLayout`, then call `CrossSectionModel.model_rebuild()`.

For further information visit https://errors.pydantic.dev/2.11/u/class-not-fully-defined
>>> 
```

**To Reproduce**

I created a virtual env with `python3 -m venv .venv` and then installed gdsfactory using `pip install -r requirements.txt` where my `requirements.txt` file had the following;
```
gdsfactory>=7.0.0
numpy>=1.24.0
git+https://github.com/gdsfactory/gf180.git
```

It looks like I ended up with `gdsfactory-9.2.2` installed.

I'm running ubuntu;
```
DISTRIB_ID=Ubuntu
DISTRIB_RELEASE=24.04
DISTRIB_CODENAME=noble
DISTRIB_DESCRIPTION="Ubuntu 24.04.2 LTS"
```

mithro, 2025-05-08:
The issue seems to be caused by installing the `gf180` package.

The following `requirements.txt` file works fine;
```
gdsfactory
```

The following `requirements.txt` file fails with the pydantic error;
```
gdsfactory
git+https://github.com/gdsfactory/gf180.git
```

sebastian-goeldi, 2025-05-08:
Likely related to pydantic 2.11. That one threw a wrench into the somewhat decent managed dependencies (at that point pydantic was >2.8 <3 or similar). Afterwards we fixed that issue and pinned pydantic more strictly. So if you either pin pydantic < 2.11 or update gdsfactory this should go away

joamatab, 2025-05-09:
Hi Tim,

That PDK isn't the one maintained under the GDSFactory organization. In our official PDKs, we always pin the GDSFactory version with both minimum and maximum bounds to ensure compatibility.

To get the correct version, you can simply run:

pip install gf180

You can also check the pinned version in the [pyproject.toml ](https://github.com/gdsfactory/gf180/blob/main/pyproject.toml)file.

Let me know if you have any issues!

Best,

mithro, 2025-05-09:
> That PDK isn't the one maintained under the GDSFactory organization.

Hrm? I was installing the PDK from `git+https://github.com/gdsfactory/gf180.git` which is the under the GDSFactory organization?

>  In our official PDKs, we always pin the GDSFactory version with both minimum and maximum bounds to ensure compatibility.

The issue is that the official PDK is forcing an older version of GDSFactory which has the pydantic error. Generally a maximum pin is a bad idea for this reason.

joamatab, 2025-05-10:
Hi Tim,

are you using a version of python supported by GDSFactory? 3.11, 3.12 or 3.13?

mithro, 2025-05-10:
Here is a minimal reproducible example -> https://github.com/mithro/gdsfactory-gf180-broken-example

I'm using Python 3.21.3 which ships with Ubuntu;
```
$ python --version
Python 3.12.3
```

Patch at https://github.com/gdsfactory/gf180/pull/87 fixes the issue.

sebastian-goeldi, 2025-05-11:
I would recommend to not unbind gdsfactory. klayout has some versions which are not backwards compatible. I would rather just have dependabot active and have it make PRs and make it bound like `gdsfactory>=9.5,<9.6`