gdsfactory/gdsfactory #3855: Pycon Sprints 2025

joamatab, 2025-05-17:
- [ ] Improve GDSFactory docs
- [ ] Improve docs for open source PDKs
    - [ ] sky130 https://github.com/gdsfactory/skywater130
    - [ ] gf180 https://github.com/gdsfactory/gf180
    - [ ] ubc-siepic
- [ ] improve physics simulator plugins https://github.com/gdsfactory/gplugins
    - [ ] Fix Meep plugin
    - [ ] Fix Lumerical plugin
    - [ ] Add SPICE sims
    - [ ] Add Quantum simulators for quantum circuits
- [ ] Build cells using multiple cores
- [ ] Improve code quality
    - [ ] Add more tests
    - [ ] Improve code coverage
    - [ ] Improve code quality
    - [ ] Improve documentation
    - [ ] Improve training materials
    - [ ] Use more ruff rules
    - [ ] Use Ty for type checking?

MatthewMckee4, 2025-05-17:
> Use Ty for type checking?

I can set this up like i did for kfactory, we will have quite a few errors initially but i can gradually fix them over time

jchanke, 2025-05-18:
add instructions for users to locally clone and develop in `README.md`. currently instructions are in the Makefile

jchanke, 2025-05-18:
convert tutorial from [jupyter notebook](https://github.com/gdsfactory/gdsfactory/archive/refs/heads/main.zip) to [marimo](https://marimo.io/)

discdiver, 2026-01-22:
Let's close this issue list and open any remaining as individual issues, @joamatab .

closed 2026-02-07 by sebastian-goeldi