gdsfactory/gdsfactory #3954: LayerMap object and gf.get_layer is kind of confusing

mithro, 2025-06-01:
The LayerMap object is an `aenum._enum.EnumType` enum. 

This makes things a bit weird because the actual int value of the layers is in no way connected to the GDS number. See the info below;
```
In [4]: gf180mcu.layers.LAYER.metal1
Out[4]: <LAYER.metal1: 10>

In [5]: gf180mcu.layers.LAYER.metal1[0]
Out[5]: 34
```

Then there is `gf.get_layer` which returns *either* a number **or** a value from the LAYER enum which I then need to convert to an actual Layer enum.
```
In [5]: gf.get_layer((0,0))
Out[21]: 131

In [26]: gf180mcu.layers.LAYER(gf.get_layer((0,0)))
Out[26]: <LAYER.pr_bndry: 131>
```

What is the correct / easy way to convert the two-tuple into a Layer object so I can get it's name?

mithro, 2025-06-01:
So far I have ended up using the following function;

```python

def gdstuple_to_layer(layer_tuple):
    """
    Convert a GDS layer tuple to a gf180mcu layer.
    
    Args:
        layer_tuple: Tuple containing (layer_number, datatype)
        
    Returns:
        Layer object from gf180mcu.layers
    """
    try:
        return LAYER(gf.get_layer(layer_tuple))
    except ValueError:
        return aenum.extend_enum(LAYER, f"Unknown {layer_tuple}", layer_tuple)

```

mithro, 2025-06-04:
Kind of related weirdness -- Typing tools think that `LAYER` takes two arguments (layer & datatype) when it actually takes a int?

See 
[image]

```
>>> help(LAYER.__init__)
Help on function __init__ in module kfactory.layer:

__init__(self, layer: 'int', datatype: 'int') -> 'None'
    Just here to make sure klayout knows the layer name.

>>> LAYER(32,0)
File ~/github/wafer-space/gf180mcu-test/.venv/lib/python3.12/site-packages/aenum/_enum.py:1848, in EnumType.__call__(cls, value, names, module, qualname, type, start, boundary)
   1833 """Either returns an existing member, or creates a new enum class.
   1834 
   1835 This method is used both when an enum class is given a value to match
   (...)   1845 the resulting class will not be pickleable.
   1846 """
   1847 if names is None:  # simple value lookup
-> 1848     return cls.__new__(cls, value)
   1849 # otherwise, functional API: we're creating a new Enum type
   1850 return cls._create_(value, names, module=module, qualname=qualname, type=type, start=start, boundary=boundary)

File ~/github/wafer-space/gf180mcu-test/.venv/lib/python3.12/site-packages/aenum/_enum.py:2314, in __new__(cls, value)
   2312         raise ValueError('%s() should be called with a value' % (cls.__name__, ))
   2313     else:
-> 2314         raise ValueError("%r is not a valid %s" % (value, cls.__name__))
   2315 else:
   2316     raise TypeError(
   2317             'error in %s._missing_: returned %r instead of None or a valid member'
   2318             % (cls.__name__, result)
   2319             )

ValueError: (32, 0) is not a valid LAYER

>>> LAYER(49)
<LAYER.nplus: 49>
```

sebastian-goeldi, 2026-09-14:
Might be natural, but conflicts with klayout internally using single integers as unique identifiers which in no way coincide with any layer or layer number likely