gdsfactory/gdsfactory #3955: Non-Manhatton connection found when using ports ROUTING

synesios, 2025-06-01:
**Describe the bug**
I want to using gf.routing to automaticlly connect two sets of  waveguides. however , I found that it works perfect on the lower side, however , non-manhantton connection comes out on the upper side, and it really confuses me about how to fix that error, and I even never know how to add .create_vinst to rewrite my code as your website shows . plz help me !

[image]

**Environment (please complete the following information):**

- [ yes ] I have reviewed the documentation, discussions and issues and found no relevant solution.
- [ yes ] I am using the latest version of GDSFactory 9.7.0, with Python 3.11

Please provide the output for the following code:

```python
# ADD OUTPUT WAVEGUIDES

for i in range(0, num2 ):
        port_to_connect = f"in{i+1}"

        if port_to_connect in fsr_in.ports:
            current_port = fsr_in.ports[port_to_connect]
            taper_in = top_device.create_vinst(gf.components.taper(length=Lt_out, width1=width_out, width2=width_s, with_bbox=True, with_two_ports=True, cross_section='strip'))
            taper_in.connect(port='o1', other=current_port)

            
            component1 = top_device.create_vinst(gf.components.straight(length = 500+200-111+15*(num2-i), width=width_s ))
            component1.connect(port='o1', other=taper_in.ports['o2'])

            component2 = top_device.create_vinst(gf.components.bend_circular(radius=100, width= width_s , angle=90+(((num2-1)/2-i)*dr/ra)))
            component2.connect(port='o1', other=component1.ports['o2'])

            top_device.add_port(name=f"intaper_out_{i+1}", port=component2.ports['o2'])
        else:
            print(f"警告: 在 FSR 组件的引用中未找到端口 {port_to_connect}。")

# ADD  OUTPUT GRATINGS
for i in range(0, num2 ):
        
        grating_out= top_device.add_ref(gra_1550_3618())
        grating_out.rotate(-45)
        grating_out.movey(-900)
        grating_out.movex(-300+100*i)
        component1 = top_device.create_vinst(gf.components.taper(length=300, width1=width_s, width2=15,with_bbox=True, with_two_ports=True, cross_section='strip'))
        component1.connect(port='o2', other= grating_out.ports['o1'])
        
        component11 = top_device.create_vinst(gf.components.straight(length = 100+15*i, width=width_s ))
        component11.connect(port='o2', other=component1.ports['o1'])

        component2 = top_device.create_vinst(gf.components.bend_circular(radius=100, width= width_s , angle=-45))
        component2.connect(port='o2', other=component11.ports['o1'])
    
        component3 = top_device.add_ref(gf.components.straight(length = ((716/8)*i+15*(num2-i)), width=width_s ))
        component3.connect(port='o2', other=component2.ports['o1'])
        
        component4 = top_device.add_ref(gf.components.bend_euler(radius=100, width=width_s))
        component4.connect(port='o2', other=component3.ports['o1'])
     
        top_device.add_port(name=f"bend_connect_to_grating_{i+1}", port=component4.ports['o1'])

        routes = gf.routing.route_bundle_sbend(
            top_device,
            [component4.ports['o1']],
            [top_device.ports[f"intaper_out_{i+1}"]],
            enforce_port_ordering=False,
            use_port_width=True,
        )
```

joamatab, 2025-06-02:
Hi,

Could you please post the code so we can try to reproduce the issue on our side?

Thanks!

synesios, 2025-06-02:
Hi, it's my pleasure to share you with my code , however ,  I don't how to send  you a personal share link because I don't want my code to be downloaded by others.  . Is there any ways in github?

sebastian-goeldi, 2025-06-16:
Hi, sorry for the long delay. I had a look at your code. The problem is that you are using `gdsfactory.components.bend_circular` fo rthe bends on the top left. You should use `gf.components.bend_circular_all_angle` instead (same for the euler bend potentially if it's anything but 90°). Reason is that the port cannot be on the ideal position and the snapping of the bend ends in non-90° cases causes an angle mismatch between port and actual polygon. The bend_all_angle delays snapping until the gds is written or at least until the cell is finished to be precise.