gdsfactory/gdsfactory #3992: Supporting `from_image` for the SVG format

abdelq, 2025-06-17:
**Is your feature request related to a problem? Please describe.**
I was hoping `from_image` would accept SVG as a format (even if it meant that it would be rasterized under the hood)

**Describe the solution you'd like**
How would you like this feature to function? What should happen when it's implemented?

- [x] I understand that, as an open-source project, the maintainers may have limited capacity to address every request. I am open to contributing by developing the feature myself or hiring someone to implement it (contact@gdsfactory.com). Otherwise, I understand that this feature may not be prioritized.

tvt173, 2025-06-18:
what is the use case? importing a logo? to a single GDS layer?

tvt173, 2025-06-18:
i see... yeah a quick and dirty way would be to go from svg -> image -> gds, but of course a much better and more efficient conversion directly from SVG to polygons should also be possible. something like [this library](https://github.com/mathandy/svgpathtools) might be handy if we want to go that route

abdelq, 2025-06-18:
> should be as easy as this https://gist.github.com/mtcomb/d691a622c55800602151 . And then add polygons from those numpy arrays (maybe they are in the wrong shape, but the principal still stands)

Thanks for the link, I was going to opt for a similar approach.