gdsfactory/gdsfactory #3993: Document virtual cells

abdelq, 2025-06-17:
`vcell` exists but is hidden and undocumented.

The same lack of documentation applies to `cell` too (with params like `with_module_name`)
https://gdsfactory.github.io/gdsfactory/_autosummary/gdsfactory.cell.html

tvt173, 2025-06-18:
agreed. @sebastian-goeldi and I were just discussing yesterday that virtual cells need to be better documented. good catch on the documentation for `cell` also. definitely deserves better documentation. i am also noticing that `ComponentAllAngle` is missing from the `Component Definition` section of the API docs. I would argue that cell and vcell should also be listed there since they are essential to cell creation. agreed?

[image]

sebastian-goeldi, 2025-06-18:
Technically `vcell` is not undocumented. It's a pure alias to kfactory's vcell. Reason being, there is nothing gdsfactory's ComponentAllAngle does on top of kfactory's VKCell, compared to Component which adds a bunch of stuff onto KCell.

https://gdsfactory.github.io/kfactory/reference/kfactory/#kfactory.KCLayout.vcell

Not that kfactory does a much better job of documenting :D