gdsfactory/gdsfactory #4009: c.add_ref() should give the possibility to provide the placement specifications

Arka-noid, 2025-06-25:
**Is your feature request related to a problem? Please describe.**
The function `add_ref` should provide support for directly specifying the placement specifications.
The add_ref represent the Instance function in KLayout, whose placement can be completely described by the coordinate (x, y), the angle, and the flip on one axis. Any possible placement can be fully described in this way.
Not providing support for rotation and mirror means that I always need to do it after
```python
c = gf.Component()
comp = gf.get_component("bend")
ref = c.add_ref(comp) # or c << comp
ref.drotate(90).dmirror_y().dmove(4, 0)
```
While this works, this always force you to 
- Create a variable to move the reference (or enclosing in parenthesis)
- A bit cumbersome to develop a custom placer function (like the `gf.read.from_yaml`) when reading from files or dataframes

**Describe the solution you'd like**
An add_ref function that supports placement specifications.
This is a wrapper I developed

```python
def add_sref(c: ComponentSpec,
             component: ComponentSpec,
             specs: Coordinate = (0., 0.),
             relative_to: Optional[Union[Coordinate, gf.Port, str]] = None,
             ) -> ComponentOrReference:

    ref = gf.get_component(component)
    inst = c << ref

    relative_to = relative_to or (0., 0.)

    if isinstance(specs, pd.Series):
        x, y = specs["X"], specs["Y"]
        a = specs.get("A")
        m = specs.get("M", False)
    else:
        x, y, *other = specs
        a = other[0] if len(other) > 0 else None
        m = other[1] if len(other) > 1 else False
    if a is not None:
        inst.drotate(a)
    if m:
        inst.mirror_y()

    if isinstance(relative_to, str):
        x0, y0 = inst[relative_to].dcenter
    elif isinstance(relative_to, gf.Port):
        x0, y0 = relative_to.dcenter
    elif isinstance(relative_to, pd.Series):
        x0, y0 = relative_to["X0"], relative_to["Y0"]
    else:
        x0, y0 = relative_to

    inst.dmove((x0, y0), (x, y))
    return inst
```

which can be used as
```
add_sref(c, comp, (4, 6))
add_sref(c, comp, (2, 6, 180))
add_sref(c, comp, (1, 4, 180, True), relative_to=(1, 2))
```
while not being a perfect solution, it is handy for complex placement specifications

- [X ] I understand that, as an open-source project, the maintainers may have limited capacity to address every request. I am open to contributing by developing the feature myself or hiring someone to implement it (contact@gdsfactory.com). Otherwise, I understand that this feature may not be prioritized.

sebastian-goeldi, 2025-06-25:
Your example can be simplified to `c.add_ref(comp).drotate(90).dmirror_y().dmove(4, 0)`. And as you see in your own example, order matters, which is what you take away if you add the placement call into the function. Therefore then you need to know how a klayout transformation applies in order (angle -> mirror -> translation).

Arka-noid, 2025-06-25:
yes, I completely agree that order matters, this is regardless of the implementation.
The order it was assumed may cover the most common use-case (you want to rotate-flip before translation) and I think it is reasonable assumption. 
You could perform other operations later in case, e.g. to rotate after translation then just apply the drotate to the output of the function.
Another option is to provide an optional parameter for the point of the rotation symmetry but I don't think it is particularly useful.

If I compare it to the documentation of the function `gf.read.from_yaml`
```python
        instances:
            name:
                component: (ComponentSpec)
                settings (Optional)
                    length: 10
                    ...
        placements:
            x: float, str | None  str can be instanceName,portName
            y: float, str | None
            rotation: float | None
            mirror: bool, float | None float is x mirror axis
            port: str | None port anchor
        connections (Optional): between instances
        ports (Optional): ports to expose
        routes (Optional): bundles of routes
            routeName:
            library: optical
            links:
                instance1,port1: instance2,port2
```
I also see that you can plug these parameters during the placements, which means you also make an assumption. You can use the same assumption for the add_ref function.

mpasson, 2025-06-26:
Totally agree! The arguments provided by the `placement` section of the yaml seem the most natural to me. Having them exposed in a single function that does the reference creation and placement would be wonderful! It would remove the necessity to give always a name to the reference every time one wants to use a port as an anchor.