gdsfactory/gdsfactory #4018: Components created with dup skipped in get_netlist

mpasson, 2025-07-01:
Hi!

I was trying the `get_netlist` function and it seems that components created with the `dup` method do not appear in the extracted netlist.

For example, the following code:

```python
c = gf.Component()
mmi = gf.c.mmi1x2()
mmi1 = c << mmi
st1 = c << gf.c.straight(length=10.0, cross_section="strip")
st2 = c << gf.c.straight(length=10.0, cross_section="strip")
mmi2 = c << mmi.dup("different name")

st1.connect("o1", mmi1.ports["o2"])
mmi2.connect("o3", st1.ports["o2"])
st2.connect("o1", mmi1.ports["o3"])
```

results in this netlist:
```yaml
instances:
  mmi1x2_gdsfactorypcompo_1f097353_0_0:
    [...]
  straight_gdsfactorypcom_f2286423_15500_625:
    [...]
  straight_gdsfactorypcom_f2286423_15500_m625:
    [...]
name: Unnamed_4
nets:
- p1: mmi1x2_gdsfactorypcompo_1f097353_0_0,o2
  p2: straight_gdsfactorypcom_f2286423_15500_625,o1
- p1: mmi1x2_gdsfactorypcompo_1f097353_0_0,o3
  p2: straight_gdsfactorypcom_f2286423_15500_m625,o1
placements:
[...]
warnings:
  optical:
    unconnected_ports:
    - message: 3 unconnected optical ports!
      ports:
      - mmi1x2_gdsfactorypcompo_1f097353_0_0,o1
      - straight_gdsfactorypcom_f2286423_15500_625,o2
      - straight_gdsfactorypcom_f2286423_15500_m625,o2
[...]
```
Is this the intend behavior? What are the requirement for a component to be picked up from get_netlist?

I had a similar problem with virtual instances, but I notices it was solved quite recently. Many thanks!

I'm running GDSfactory 9.9.4 on python 3.12.

sebastian-goeldi, 2025-07-01:
Hello,

This is a more philosophical question/problem about `get_netlist`. Currently it will skip anything not decorated with `@cell` or `@vcell` (on the cell part, not instance part). You can see the code here:

https://github.com/gdsfactory/gdsfactory/blob/main/gdsfactory/get_netlist.py#L267

If you want to use the duplicate, I would suggest to pack it again with a decorator.

sebastian-goeldi, 2025-07-11:
I think we might did that in order to make vcells pass initially

sebastian-goeldi, 2025-07-11:
I believe that would currently violate the scheme used by get_netlist and it would not be compatible with the yaml placer anymore. I'd be hesitant to introduce such a change

tvt173, 2025-07-13:
may i ask why you tend to have so many cells like this? maybe there is a better way to do what you are doing...

i would also tend to say that if you are just making a function like this once centrally and reusing it everywhere, i wouldn't care too much how ugly the function itself is, as long as the end-user doesn't have to be concerned with it... 

a shorter but more dangerous way to do it would be to just explicitly set `c.function_name`, which would hack the implementation of `get_netlist()` directly, but it would leave you vulnerable to future changes in gdsfactory, as the general assumption is that "proper" cells are generated with an `@cell`-decorated function

tvt173, 2025-07-13:
@mpasson I think this [PR](https://github.com/gdsfactory/gdsfactory/pull/4030) should resolve your issue. I would still avoid using cells which are undecorated, if at all possible.

PR #4030 (2025-07-14): Netlist undecorated insts
@mpasson pointed out that undecorated cells do not get netlisted when using the `get_netlist()` function. I think that this is dangerous behavior. This PR netlists such cells, using the component name rather than `function_name`, when it is not available. There is a warning when this happens.

Also, I fixed two generic cells which were violating best practices, found by this fix :)

Fixes #4018

## Summary by Sourcery

Enable netlisting of cells without function_name by using their component name, refactor ring_crow creation into a decorated helper cell, flatten termi [... cut]
closed 2025-07-14 by joamatab