gdsfactory/gdsfactory #4021: route_bundle of different widths

dsus4wang, 2025-07-05:
[image]

Hello guys.
I have several wires of different widths that need to be connected using a function like route_bundle, but I see that route_bundle only allows one width. Is there any other way to achieve connections of different widths?

thanks.

tvt173, 2025-07-11:
i don't think this is currently possible, right @sebastian-goeldi ?

although i could imagine you could do it if you route a bundle with a sufficiently large separation, and then recreate each individual route with your width of choice, using the backbones output from the bundle router

sebastian-goeldi, 2025-07-11:
Hmm, that depends on what you call not working. I believe the router itself should be able to do it. But the "frontend" function cannot handle different bend radii. And gdsfactory doesn't expose different route widths

AndreaWCKim, 2025-07-31:
I made a width variation with route_bundle using one additional layer.
The start points and end points of the straight lines were filtered, 
and the (taper + straight + taper) components are placed between start and end point of each straight line.

[image]

[image]

[image]