gdsfactory/gdsfactory #4047: Effective Methods for Drawing Shared Ground Paths

cheng1214, 2025-07-22:
Hi,

I was wondering if there's a good method for drawing the shared ground path, similar to the figure below.

[image]

Thanking you!

sebastian-goeldi, 2025-07-22:
Not sure I understand what exactly you are trying to achieve. If it's purely about the gdsii concept of path, you can do that with the klayout object:

`c.shapes(gf.get_layer(layerspec)).insert(gf.kdb.DPath([gf.kdb.DPoint(x1,y1), gf.kdb.DPoint(x2,y2), ...], path_width))`

I generally would highly advise against using path objects directly in the gds, especially if you are doing stuff like here where you have a self-intersecting path. It is not clearly defined by the gdsii spec how a path should be drawn from the path coordinates (this concerns corners and especially the endings of the path). Therefore, it is highly advisable to convert it to a polygon and use the polygon instead, unless your structure really requires a path and you are certain as to how the target-tool will handle it (I think some e-beam tools for example use paths).

cheng1214, 2025-07-23:
Thank you for your helpful response! 
Additionally, I was inquiring whether there is a dedicated routing function designed to handle such shared ground paths. Otherwise, manually drawing these paths would be quite cumbersome.