gdsfactory/gdsfactory #4066: Routing conflict error when using add_`fiber_array` with multiple ports

cheng1214, 2025-07-29:
Hi!

I'm encountering a weird error when using add_fiber_array to route multiple ports to grating couplers.

In my first case, I have two ports ('o5' and 'o5') that can be successfully routed to grating couplers individually (see first image below, where the dashed area is 'bbox' of the component ). 

[image]

However, when I try to include both ports in add_fiber_array for simultaneous routing, I get the following error:

`ValueError: The sorting algorithm currently doesn't support if multiple bundles are conflicting with each other. Offending bundle at starting port positions box.left=-28250,box.bottom=-20860,box.right=156880,box.top=37609[dbu]
`

What's confusing is that these two ports are far apart and should easily route to separate grating couplers without conflict.

Interestingly, in a second similar case with two different ports (shown in the second image), the routing works perfectly fine with add_fiber_array.

[image]

Could anyone help me understand why these ports might be considered conflicting in the first case, and suggest how to resolve this issue? I would greatly appreciate any guidance on this matter.

Thank you in advance!

cheng1214, 2025-07-30:
I've identified the root cause of the issue. The problem occurs because I enabled the auto-taper option in the add_fiber_array method, which automatically inserted a taper structure at port 9. This modification extended the port's x-coordinate beyond the component's original bounding box, causing it to fall outside the grouping area shared with port 5. As a result, the routing algorithm failed to group these ports together in the same bounding box, ultimately leading to a mismatch between the number of start bundles and end bundles during the routing process.

sebastian-goeldi, 2025-07-30:
Great that you were able to figure it out! 

Yeah, the algorithm can be a bit confusing with the error messages. As a rule of thumb, the implementation only works well if the end_ports of the bundle are on a line like structure (or can at least be alinged and spearated into sub-bundles which can be handled individually), this especially applies to when using the auto sort algorithm.

Maybe this will get better once we have proper a*routing. So stay tuned ;)