gdsfactory/gdsfactory #4086: Ability to flatten generated route into a single polygon

abdelq, 2025-08-07:
**Is your feature request related to a problem? Please describe.**
I have a custom route made using `route_bundle` that uses `steps`, I'd like to flatten the whole thing to a single cell in the hierarchy of the GDS.

**Describe the solution you'd like**
Could be another flag in `route_bundle`?

- [x] I understand that, as an open-source project, the maintainers may have limited capacity to address every request. I am open to contributing by developing the feature myself or hiring someone to implement it (contact@gdsfactory.com). Otherwise, I understand that this feature may not be prioritized.

joamatab, 2025-08-07:
Why not just run 

C.flatten()

after placing the route?

sebastian-goeldi, 2025-08-07:
that, or you add the route to a separate cell. Nothing prevents you from using a different cell to store the routes (and flatten if you want to in there)