gdsfactory/gdsfactory #4087: Unexpected behaviour when using route_bundle's steps

abdelq, 2025-08-07:
**Describe the bug**
Using `route_bundle` with `steps`, going down and then to the right. For reasons that escape me, I have to set a `dx` of a certain value (don't know how to compute it) to make it go down in a straight line.

What I want:
[image]

What I get:
[image]

**To Reproduce**
```
import gdsfactory as gf

c = gf.Component()

r1 = c << gf.c.rectangle((50, 50))
r2 = c << gf.c.rectangle((50, 50))
r2.movex(250)

gf.routing.route_bundle(
    c,
    [r1.ports["e4"]],
    [r2.ports["e4"]],
    route_width=10,
    layer="WG",
    auto_taper=False,
    steps=[
        {"dx": 30, "dy": -100},  # Remove dx to get the odd behaviour
        {"dx": 50},
        {"dy": 50},
        {"dx": 100},
        {"dy": -50},
        {"dx": 30},
    ],
)

c.show()
```

**Expected behavior**
If I'm setting `dy` only, I should expect a straight line.

**Environment (please complete the following information):**

- [x] I have reviewed the documentation, discussions and issues and found no relevant solution.
- [x] I understand that this is an open-source project, and maintainers may not have the resources to address every issue. I am prepared to contribute by fixing the issue myself or hiring someone to do so, if needed, and accept that the issue may not be resolved otherwise.
- [x] I am using the latest version of GDSFactory, with Python 3.11, 3.12, or 3.13.

Please provide the output for the following code:

```python
                                                                                                 Modules
┏━━━━━━━━━━━━━━━━┳━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━┳━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━┓
┃        Package ┃ version                                              ┃                                                                                                                          Path ┃
┡━━━━━━━━━━━━━━━━╇━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━╇━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━┩
│         python │ 3.12.6 (main, Sep  9 2024, 22:11:19) [Clang 18.1.8 ] │                                .venv/bin/python │
│     gdsfactory │ 9.12.1                                               │   .venv/lib/python3.12/site-packages/gdsfactory │
│         devsim │ not installed                                        │                                                                                                                               │
│        femwell │ not installed                                        │                                                                                                                               │
│ gdsfactoryplus │ not installed                                        │                                                                                                                               │
│       gplugins │ not installed                                        │                                                                                                                               │
│       kfactory │ 1.12.1                                               │ ['.venv/lib/python3.12/site-packages/kfactory'] │
│         lumapi │ not installed                                        │                                                                                                                               │
│           meep │ not installed                                        │                                                                                                                               │
│           meow │ not installed                                        │                                                                                                                               │
│            ray │ not installed                                        │                                                                                                                               │
│            sax │ not installed                                        │                                                                                                                               │
│         tidy3d │ not installed                                        │                                                                                                                               │
│          vlsir │ not installed                                        │                                                                                                                               │
└────────────────┴──────────────────────────────────────────────────────┴───────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────────┘

```

abdelq, 2025-08-07:
This is very light on details, but could be related: https://github.com/gdsfactory/gdsfactory/issues/3948
I also think my comment here might be relevant to the whole `route_bundle` behaviour: https://github.com/gdsfactory/gdsfactory/issues/4002#issuecomment-3162020878

Unrelated, but it would be good if `gf.config.print_version_plugins()` doesn't leak the whole path.

sebastian-goeldi, 2025-08-13:
in gf9 waypoints are treated like gates. The routes must pass the first and last gate with a straight. therefore your example will cause the following problem:

[image]