gdsfactory/gdsfactory #4105: recurse gdsfactory cells

joamatab, 2025-08-14:
can we add this by default

c = gf.Component(base=kcell.base)

@sebastian-goeldi

sebastian-goeldi, 2025-08-14:
Do where by default? Not following

joamatab, 2025-08-15:
yes, sorry, when we recurse gdsfactory cells we get kfactory objects instead, is there a way to fix this behavior?

sebastian-goeldi, 2025-08-15:
Not sure which one you are talking about. Could you point me to which one you mean?

The instances should be the same in gdsfactory and kfactory as far as I know and there are no other recursive iterators which are directly accessible by kfactory.

mpasson, 2025-08-27:
I'm not sure if this is related, but I was indeed wondering if it is possible to go back to the original component from an instance. For example:
```python
top = gf.Component('top')
c = gf.Component('c')
top << c

for inst in top.insts:
    inst # this is a kfactory Dinstance
    inst.cell # this is a kfactory DKcell 
```
is there now any way to get the component back? Something that can be used in a `if inst.component is c` statement, for example?

sebastian-goeldi, 2025-08-27:
sure. `gf.Component(base=inst.cell)` will give you back the component. But for here, there is no easy way to make the Instance give component back automatically from this. A DKCell is very close to a Component already though