gdsfactory/gdsfactory #4112: Trouble with synchronizing KLayout and GDSFactory

juanfmorcillo, 2025-08-21:
Hello, I'm new to this software and I'm having some problems with integrating KLayout. I'm searching for the GDSFactory plug-in in KLayout at 'Tools/Manage Packages'. I click on 'gdsfactory 9.12.2' and apply. However, an error arises at fetching the spec file for the package. I would kindly require some help with this issue. Thanks in advance!

[image]

joamatab, 2025-08-21:
What version of klayout do you have?

Shouldn't you make this issue in KLayout?

juanfmorcillo, 2025-08-21:
I'm working with v.0.28.3. 

I'll post it in /klayout as well. Thanks!