gdsfactory/gdsfactory #4152: (question) How can i install Gdsfactory correctly ?

optic-yellowfly, 2025-09-10:
I follow the below process to install the gdsfactory

[image]

but something went wrong when i  run the code, the wrong message is

ModuleNotFoundError: No module named 'gdsfactory'

I found that the installed package is gdsfactory_install, not gdsfactory, but if i modify the code as "import gdsfactory_install as gf", the wrong message still occured, which is

 AttributeError: module 'gdsfactory_install' has no attribute 'Component'

MatthewMckee4, 2025-09-10:
Where is this notebook snippet from?

ZijunChen147, 2025-09-19:
Having the same issue with installation...

joamatab, 2025-09-20:
Hi,

we can't help you if you don't show the screenshot of the message that you get

Have you tried the developer install?
https://gdsfactory.github.io/gdsfactory/developer.html

luomeng007, 2025-11-05:
When I installed, I also meet some problem. after I run `gfi install`. I got error:

[image]

What should I do next to overcome this? 

Besides, I find the method for using `pip install gdsfactory` is invalid now.