gdsfactory/gdsfactory #4182: GDSFactory takes longer than Nazca Design to generate a GDS file

afhinestrosa, 2025-09-22:
I have been using the Nazca Design package for 3+ years now. I typically use it for generating photonic circuit layouts. More specifically, I code Python scripts that generate waveguide gratings with arbitrary geometry per period. My cells are difficult to decorate (hash), because every one is different from the rest. When building complete layouts for fabrication, the GDS files can take up to half an hour for generating.

Recently, I found GDSFactory appealing, due to its selling point as a fast and efficient package. However, when I got to compare it to the Nazca Design package, the latter provided me with better results. I implemented the same Python code using both packages. The code generates a GDS file with a sidewall Bragg grating of 1500 periods, each period with a different `sideblock_width`. Nazca Design took around 1 second to generate this layout, whereas GDSFactory took 20 seconds. I am using the latest version of GDSFactory (9.15.1) and the 0.5.13 version of Nazca Design.

I am confident that I am doing something wrong, so I would really like that someone points me in the right direction. Both files are attached for reference. I will ask any questions and thank you in advance.

[SidewallGratingGDSFactory.py](https://github.com/user-attachments/files/22462338/SidewallGratingGDSFactory.py)
[SidewallGratingNazcaDesign.py](https://github.com/user-attachments/files/22462339/SidewallGratingNazcaDesign.py)

afhinestrosa, 2025-09-22:
Thank you a lot, @tvt173 !

I did as you suggested, but using the `shapely` Python package. The time got reduced to 0.22 seconds for GDSFactory and 0.34 seconds for Nazca Design. This will for sure make the difference for my projects. Attached you can find the Python scripts.

@joamatab, there is no need to convince me about GDSFactory being faster than its competitors, there is plenty of proof available :) However, never before I had attempted to solve this specific problem. The code I attach now is flawless, but I am afraid I can not follow the guide you sent me. My arrays change geometry per period, so I believe that I am forced to draw them individually as polygons. This will also happen to people that work with e.g. metasurface lenses. If you find any better solution let me know, and thank you anyway for the early answers!

[SidewallGratingGDSFactoryWithPolygons.py](https://github.com/user-attachments/files/22468925/SidewallGratingGDSFactoryWithPolygons.py)
[SidewallGratingNazcaDesignWithPolygons.py](https://github.com/user-attachments/files/22468924/SidewallGratingNazcaDesignWithPolygons.py)

afhinestrosa, 2025-09-22:
Thank you a lot, @sebastian-goeldi .

1. Totally true, and bad from my side. In any case, it will now always be a linspace, sometimes it is an array with arbitrary values. In which case, I will follow your advice: make a dictionary, put a name, and pass the name to the function instead of the array of values. Pretty useful comment, thank you again.
2. About this, I already tried what you are mentioning and it did not speed up the code, maybe I am wrong. Will try again.
3. I will definitely try this, and I must say I really tried from the start but due to my lack of experience with GDSFactory I did not find the way of using the DBoxes. Thank you for this suggestion too.

I will upload the new code once it's finished, I hope we reduce even more the time it takes to draw :)

sebastian-goeldi, 2025-09-22:
I didn't profile the code. But from your description, I am pretty sure having a mapping available where you only pass the key to the function is the culprit, as even millions of shapes only make a dent in the runtime and don't blow it up like crazy usially, especially for box like structures. So yes, maybe I should have stated that clearer. I assume that once the big culprit is gone, 2. and 3. will only make an impact in the range of 5% or so from personal experience.